dhm2013724/yolov2_xilinx_fpga #3: Hello, your work is very good, but I have some questions, can you give some advice?

wu-y, 2018-11-29:
Hello, I am recreating your code in vivado_hls2017. It takes 45 minutes to test a picture. What is the reason? I want to reproduce your code on the ZCU102. Can you give some advice?

dhm2013724, 2018-11-29:
> Hello, I am recreating your code in vivado_hls2017. It takes 45 minutes to test a picture. What is the reason? I want to reproduce your code on the ZCU102. Can you give some advice?

C Simulation always takes too much time, and I have warned others when I uploaded the testbench.   There is no problem. I don't suggest you evaluate this design in ZCU102, because it is not designed to so high-level FPGA chip, and many architectures and considerations are not suitable. Maybe you can try https://github.com/Xilinx/CHaiDNN. which is designed for MPSOC

wu-y, 2019-02-16:
Hello, I found that you are using the normal transfer mode when transferring data. If you use DMA transfer, the speed is faster; you now have 2*32 parallelism due to the board, if you don't consider resource issues. To improve parallelism, where do I need to make changes?

dhm2013724, 2019-02-25:
> Hello, I found that you are using the normal transfer mode when transferring data. If you use DMA transfer, the speed is faster; you now have 2*32 parallelism due to the board, if you don't consider resource issues. To improve parallelism, where do I need to make changes?

I have implemented the 4*32 parallelism in Zedboard. If you want to increase the parallelism, generally there are two ways to achieve it. First, try to decrease the data precision, and you will have more resoures to design more mul-add unit. Second, You can partially or completely unroll the 2-3 dimension of six-dimensional convolution to design computing arrays with higher utilization or higher parallelism. I have just partially unrolled 2 dimension. In addition, you can try other optimizations as well, like Winograd.

wu-y, 2019-03-06:
你好，我在你设计的基础上实现了目标检测，但是有以下几个问题，希望你能答疑：
（1）在cnn.cpp中定义了一个 INTER_WIDTH =19，是如何确定的，具体作用是啥
（2）因为我只实现了单类目标检测，修改了对应的.cfg文件，和网络结构，但是在仿真过程中，我发现图像上会标注很多框，我调整了阈值，也没有用。
（3）按照你的步骤，我对weights和bias都进了量化，但是input和output该如何量化，该如何从darknet框架中提取出来。
（4）我对coco.weights进行了量化，weightQ值和你提供的是一致的，但是量化以后的结果和你提供的量化结果完全不同，是你对数据做了二次处理吗？

closed 2019-05-30 by wu-y