dhm2013724/yolov2_xilinx_fpga #4: 您好，我使用zynq进行仿真，那几个bin文件有些疑问请教您

17677174975, 2018-12-27:
代码中有五个bin文件：
其中 buf的有两个，Weight_buf（yolov2_w_reorg_bn_ap16_short16.bin）、Beta_buf ( yolov2_b_ap16_short16.bin ) ，这两个文件是什么文件，有什么用呢？
其中还有三个，inputQ（yolov2_bn_input_maxQ_24.bin ）、weightQ（yolov2_w_reorg_bn_ap16_maxQ_23.bin）、betaQ（yolov2_b_ap16_maxQ_23.bin），这三个文件是什么文件，有什么作用呢？

yolo的权重文件只有一个，就是 *.weight，如（yolov2.weights），上面的五个文件跟yolo的*.weight是什么关系呢？

wu-y, 2019-01-03:
这个你可以看一下参考文献【6】，inputQ，weightQ，betaQ，都表示定点数中小数位置，Weight_buf和beta_buf是网络的权值参数。我理解的是weight是卷积核的参数，beta应该是BN层的参数

dhm2013724, 2019-01-07:
> 代码中有五个bin文件：
> 其中 buf的有两个，Weight_buf（yolov2_w_reorg_bn_ap16_short16.bin）、Beta_buf ( yolov2_b_ap16_short16.bin ) ，这两个文件是什么文件，有什么用呢？
> 其中还有三个，inputQ（yolov2_bn_input_maxQ_24.bin [...]

最近在赶deadline，时间比较紧，前面Weight_buf和Beta_buf对应的就是weight和bias，只是这边已经把每层对应的BN也合并进去了，是先合并进去得到的float32位的Weight和Bias，然后再对其进行动态定点16位的量化，后面是对应的量化参数（就是16位定点的时候小数点放哪里），分别对应输入输出、Weight和Bias，所有的代码应该会在最近全部放在github上，从如何从Darknet取参数到最终优化的都有，之前忙着做实验，目前只要写论文了，应该会每天上来写一写，可能会重新开个project，到时候会在首页贴个链接

wu-y, 2019-02-21:
Hello, I added input parallelism to your design, using the weights and offset files you provided to test, but not the target. Is there a problem in my design, or do other processing when storing your weight file after storage. I can't read parameters correctly when I change the degree of parallelism

dhm2013724, 2019-02-25:
> Hello, I added input parallelism to your design, using the weights and offset files you provided to test, but not the target. Is there a problem in my design, or do other processing when storing [...]

Yes, actually it still need weight re-arrangement in order to increase the burst length of AXI transmission. Tonight, I will try to update the PYNQ version first. The full process will be described in another project. This project is only responsible for PYNQ.

wu-y, 2019-02-28:
Hello, I quantified the weights and bias according to the 【6】, but how to quantify the input and output, the two 16-bit numbers are multiplied, is 32 bits, and what rules are quantized to 16 bits.

Thilanka97, 2019-05-15:
@Johnny-Lin @dhm2013724 @wu-y 
Hey
I am trying to understand how this implementation works and I do not understand what weight offset and beta_offset is. Also What it does?

Also is inputQ, outputQ, weightQ and betaQ mean quantized input, quantized output, quantized weight and quantized beta? do you input both quantized weight, beta files and 32fp weight, beta files to the design? (I am confused because there are number of weight and beta files) what is the difference between weight and weightQ? how are they used in the design ?

Also what is Tr and Tc? and mloops, nloops,rloops and cloops?

I am still a beginner could you briefly tell me what these mean in the code? Atleast some of them. would be a great help :)
Thanks in advance!

evanlee2021, 2021-11-15:
> Yes, actually it still need weight re-arrangement in order to increase the burst length of AXI transmission. Tonight, I will try to update the PYNQ version first. The full process will be described [...]

您好，请问您这个项目中提到的有完整过程的另一个project在哪里呢，能麻烦您附上链接吗？

calvinee, 2023-02-04:
请问这里如何并入每层对应的BN参数？

TSeminent, 2023-02-18:
> 您好，请问您这个项目中提到的有完整过程的另一个project在哪里呢，能麻烦您附上链接吗？

同问