dhm2013724/yolov2_xilinx_fpga #5: can this code be used with other boards ?

Thilanka97, 2019-01-09:
hey @dhm2013724 thank you for uploading your code. as you said this runs at a speed of 1fps on pynq ryt? do u think that it is possible to make it run on pynq real time with tiny-yolo (if you further optimize?) Also can this be run on any other xilinx board other than pynq ? 

Thanks in advance !

dhm2013724, 2019-01-09:
> hey @dhm2013724 thank you for uploading your code. as you said this runs at a speed of 1fps on pynq ryt? do u think that it is possible to make it run on pynq real time with tiny-yolo (if you further optimize?) Also can this be run on any other xilinx board other than pynq ?
> 
> Thanks in advance !

For YOLOv2, I have reached my goal that 1 FPS, and maybe it can be further optimized to 1.5FPS in Zedboard.  And for YOLOv2-Tiny, I also reached about 4 FPS with the same accelerator.  I think that if you use 8-bit precision or lower data-precision, you can get real-time. Off course, if you use other optimization methods such as FFT, Winograd, maybe you can get better performance only if the source can be met. This accelerator can be run on other xilinx board, because it is written in HLS. If you can find the chip in Vivado HLS, you can easily realize it.

Thilanka97, 2019-01-10:
@dhm2013724 thnak you for the reply. so can u achieve 1fps with the uploaded code ? Zedboard is a beginner zynq board right ? can you use a code written in python with zynq boards ? Also would you suggest a bigger board which can be used with this code ? cause i am trying to make it work real time. 

Thanks in advance!

Thilanka97, 2019-01-17:
@dhm2013724 if you are using 8-bit data precision, then do you have to find a set of weights which are trained to be 8-bit precision or can you use the floating point weights and convert them into 8 bit weights and activations in the code ?

dhm2013724, 2019-01-18:
> @dhm2013724 if you are using 8-bit data precision, then do you have to find a set of weights which are trained to be 8-bit precision or can you use the floating point weights and convert them into 8 bit weights and activations in the code ?

Oh, I have not found the 8-bit weight. And I also tried to convert float-32 weight to fixed-8 directly,   but the precision is very bad. I think that if you want to use fixed-8 precision, you need to retrain your model. Recently, I will upload the float version and new fixed-16 version and the evaluation between different optimization methods. And the whloe design flow and source code will be available in another project. I'm not very familiar with Python so that this project may update slowly...... : (

dhm2013724, 2019-01-18:
> @dhm2013724 thnak you for the reply. so can u achieve 1fps with the uploaded code ? Zedboard is a beginner zynq board right ? can you use a code written in python with zynq boards ? Also would you [...]

I think that if you want to achieve real-time(24FPS?) with YOLOv2-Tiny, you at least  have to get the 4-bit precision and try to use the winograd fast convolutional algorithms in PYNQ or Zedboard. Off course, you can use ZC706 or ZCU102, it will be easier to achieve your goal.

Thilanka97, 2019-01-18:
> > @dhm2013724 if you are using 8-bit data precision, then do you have to find a set of weights which are trained to be 8-bit precision or can you use the floating point weights and convert them into 8 bit weights and activations in the code ?
> 
> Oh, I have not found the 8-bit weight. And I also tried to convert float-32 weight to fixed-8 directly, but the precision is very bad. I think that if you want to use fixed-8 precision, you need to retrain your model. Recently, I will upload the float version, new fixed-16 version, and the evaluation between different optimization methods. The whloe design flow and source code will be available in another project. I'm not very familiar with Python so that this project may update slowly...... : (

@dhm2013724 okayy I am looking forward for your new project. If it is not in python then is it in c? 

ohh so if you want to get 8 bit precision then we need to train ? train on fpga? if i am able to find a pre trained tiny yolo 8 bit weight set thats fine right ? then i do not need to train right ?

Thilanka97, 2019-01-18:
> I think that if you want to achieve real-time(24FPS?) with YOLOv2-Tiny, you at least have to get the 4-bit precision and try to use the winograd fast convolutional algorithms in PYNQ or Zedboard. [...]

not exactly 24fps, atleast above 15fps. ZC706 and ZCU102 are too expensive for me. I have a limited budget. thats why i have to go with zedboard. can you use python on zedboard with pync image on it. or did u use c to program zedboard ?

dhm2013724, 2019-02-25:
> not exactly 24fps, atleast above 15fps. ZC706 and ZCU102 are too expensive for me. I have a limited budget. thats why i have to go with zedboard. can you use python on zedboard with pync image on [...]

This project is about PYNQ, I have create another project about how to map YOLOv2 to Zedboard with SDK... You know. The Spring Festival has just passed, it's a long holiday. So...The project will be updated shortly.

Thilanka97, 2019-02-25:
@dhm2013724
Hey thank you! Btw have you uploaded that project to github ? If so could you please provide the link?

Thilanka97, 2019-02-27:
> @dhm2013724
> Hey thank you! Btw have you uploaded that project to github ? If so could you please provide the link?

@dhm2013724 I meant the map YOLOv2 to Zedboard, is this project available on github? Please share the link if it is already uploaded. 

Thank you so much. Your work helps alot. Really appreciate

Thilanka97, 2019-02-28:
@dhm2013724 hey btw which version of vivado did you use? free webpack version or Vivado HL Design Edition or Vivado HL System Edition?

Thanks in advance!

dhm2013724, 2019-03-01:
> @dhm2013724 hey btw which version of vivado did you use? free webpack version or Vivado HL Design Edition or Vivado HL System Edition?
> 
> Thanks in advance!

I ues Vivado HL System Edition 2018.2 in this project. You can also use Vivado 2017.2 or 2017.4 that I have tried. In few days, the project for mapping YOLOv2 to Zedboard will be updated. I'm drawing related module diagrams, it's very time-comsuming...

Thilanka97, 2019-03-04:
@dhm2013724 thanks for the reply. is the project for mapping YOLOv2 to Zedboard is also based on HLS ?

Thilanka97, 2019-03-08:
> I ues Vivado HL System Edition 2018.2 in this project. You can also use Vivado 2017.2 or 2017.4 that I have tried. In few days, the project for mapping YOLOv2 to Zedboard will be updated. I'm [...]

@dhm2013724 a quick question about Vivado HL System Edition. Do you need to have a license file for system edition, need to purchase or do educational institutes and universities get it ? 

Also to to use HLS with Zedboard also we can use the same vivado system edition right ?

Thanks in advance!

Thilanka97, 2019-03-08:
@dhm2013724 Do we need SDSOC also to use hls in zedboard ?

SoumyaDeshapnde, 2019-10-03:
Can this code be used on Zybo Z7 (Zynq-7000 ARM/FPGA SoC)?