dhm2013724/yolov2_xilinx_fpga #9: error in----

MonalPatel, 2019-02-25:
**I didnt find this file so i got errors**
#===============================================
#   yolov2 weight and bais copyto memory
#==============================================
params_wight = np.fromfile("yolov2_w_reorg_bn_ap16_short16.bin", dtype=np.uint32)
np.copyto(weight_base_buffer, params_wight)
print("yolov2_weight copy ok\n")

params_bais = np.fromfile("yolov2_b_ap16_short16.bin", dtype=np.uint32)
np.copyto(bate_base_buffer, params_bais)
print("yolov2_bais copy ok\n")

dhm2013724, 2019-02-25:
> **I didnt find this file so i got errors**
> #===============================================
>
> # yolov2 weight and bais copyto memory
> #==============================================
> params_wight = [...]

You can get YOLOv2's weight and bias from the URL in /hls repo.

MonalPatel, 2019-02-25:
yes but I am not able to download as its is baidu protected..
so can you please send it to me separately at this email address?
email id: monalrakeshpatel@gmail.com

dhm2013724, 2019-02-25:
> yes but I am not able to download as its is baidu protected..
> so can you please send it to me separately at this email address?
> email id: [monalrakeshpatel@gmail.com](mailto:monalrakeshpatel@gmail.com)

I have sent it. Good luck! ^_^

Thilanka97, 2019-04-11:
@MonalPatel @dhm2013724 Hey,
I am also facing the same difficulty. Can either one of you kindly send the weights ans bias files to me as well? 
My email: st119484@ait.asia 

Thanks in advance!