dhm2013724/yolov2_xilinx_fpga #10: Zedborad版本

zhangqipku, 2019-03-02:
大佬，请问方便加一下您的微信么  我的微信是c578232360  能不能把Zedborad更新一版，最近也忙着做在FPGA上做YOLO的加速，想请教请教问题 多谢~

zhangqipku, 2019-06-04:
Can softmax.so recompile a 64bit version?I want to port to the zcu102 development board.But the problem is"OSError: ./softmax.so: wrong ELF class: ELFCLASS32". Thank you for uploading a 64bit softmax.so.

zhangqipku, 2019-06-04:
Can softmax.so recompile a 64bit version?I want to port to the zcu102 development board.But the problem is"OSError: ./softmax.so: wrong ELF class: ELFCLASS32". Thank you for uploading a 64bit softmax.so. My emails is : 578232360@qq.com

clancylea, 2019-06-04:
@zhangqipku I have uploaded the softmax.c and softmax_a53_64bit.so.

zhangqipku, 2019-06-10:
Thank you for providing softmax_a53_64bit.so.
When I use the same method to port to the zcu102 development board.  I changed the base address of YOLO in jupyter. Although it works, but it doesn't frame the target. Should I modify the code in HLS to complete it? I want to know what the problem is? Thank you very much for your help

clancylea, 2019-06-10:
> Thank you for providing softmax_a53_64bit.so.
> When I use the same method to port to the zcu102 development board. I changed the base address of YOLO in jupyter. Although it works, but it doesn't [...]

Do you used the xlnk.cma_array to mmap memory？what do you mean changed the base address？Don't need to modify the code in HLS.

zhangqipku, 2019-06-19:
Yes I used the xlnk.cma_array to mmap memory. 
I have encountered several problems：
1.In vivado，critial warning: Excluded Address Segments HP0_LPS_OCM
2.In jupyternotebook，When I use np.copyto, the system crashes.

zhangqipku, 2019-06-19:
Maybe ZCU102 block design is different from pynq. Thank you for uploading a BD of ZCU102.
Thank you~Thank you~Thank you

closed 2019-12-27 by dhm2013724