dhm2013724/yolov2_xilinx_fpga #11: can this IP can run with other overlay?

Urjitmodhia, 2019-03-07:
hey @dhm2013724 ; I really appreciate your work as it is really amazing.
I am trying to implement your IP in my overlay which includes several another IPs. but the issue is when i run your code in my overlay it shows some errors in the second last code in .ipynb which I dont know how to resolve it. can you please help me with that?

the issue is as follows:-
===================================================================
Open pictrue success!
pictrue size: (640, 424)
pictrue mode: RGB
yolov2_image copy ok

0 conv
---------------------------------------------------------------------------
KeyboardInterrupt                         Traceback (most recent call last)
<ipython-input-19-674dc39f1ab6> in <module>()
         6 
         7 img_out = frame_in
----> 8 yolo_meminout(frame_in,img_w,img_h,img_out)
         9 img_out

<ipython-input-18-532a3d86bdd0> in yolo_meminout(frame_in, img_w, img_h, frame_out)
        31     region_buff = np.zeros((73008,), dtype=np.float32)
        32 
---> 33     yolo_fpga(img_base_buffer,region_buff)
        34     end_time = time.time()
        35     fpga_process_time = end_time - start_time

<ipython-input-10-2d57539a38b2> in yolo_fpga(img_base_buffer, region_buff)
        53                 mLoops,nLoops,rLoops,cLoops,0,
        54                 inputQ[offset_index],inputQ[offset_index+1],weightQ[offset_index],betaQ[offset_index],
---> 55                 WEIGHT_BASE,BETA_BASE)
        56 
        57 

<ipython-input-8-984fdf343523> in YOLO__Init_EX(In_Address, Out_Address, Weight_offset, Beta_offset, InFM_num, OutFM_num, Kernel_size, Kernel_stride, Input_w, Input_h, Padding, IsNL, IsBN, TM, TN, TR, TC, mLoops, nLoops, rLoops, cLoops, LayerType, InputQ, OutputQ, WeightQ, BetaQ, WEIGHT_BASE, BETA_BASE)
        53 
        54     while True:
---> 55         ap_idle =  (mmio.read(XYOLO2_FPGA_CTRL_BUS_ADDR_AP_CTRL)>>2)&0x01
        56         if(ap_idle):
        57             break

/usr/local/lib/python3.6/dist-packages/pynq/mmio.py in read(self, offset, length)
       137 
       138         self._debug('Reading {0} bytes from offset {1:x}',
--> 139                     length, offset)
       140 
       141         # Read data out

KeyboardInterrupt: 
===================================================================

the last line shows keyboard interrupt which was done by me as it was taking so much time to run the code (24 min)

please help me regarding this.....

liutongxue123, 2019-05-16:
I am also modifying this code. Can I communicate with you?

liutongxue123, 2019-05-16:
> hey @dhm2013724 ; I really appreciate your work as it is really amazing.
> I am trying to implement your IP with creating a new overlay. but the issue is when i run your code in my overlay it shows some errors in the second last code in .ipynb which I dont know how to resolve it. can you please help me with that?
> 
> # the issue is as follows:-
> Open pictrue success!
> pictrue size: (640, 424)
> pictrue mode: RGB
> yolov2_image copy ok
> 
> ## 0 conv
> KeyboardInterrupt Traceback (most recent call last)
> in ()
> 6
> 7 img_out = frame_in
> ----> 8 yolo_meminout(frame_in,img_w,img_h,img_out)
> 9 img_out
> 
> in yolo_meminout(frame_in, img_w, img_h, frame_out)
> 31 region_buff = np.zeros((73008,), dtype=np.float32)
> 32
> ---> 33 yolo_fpga(img_base_buffer,region_buff)
> 34 end_time = time.time()
> 35 fpga_process_time = end_time - start_time
> 
> in yolo_fpga(img_base_buffer, region_buff)
> 53 mLoops,nLoops,rLoops,cLoops,0,
> 54 inputQ[offset_index],inputQ[offset_index+1],weightQ[offset_index],betaQ[offset_index],
> ---> 55 WEIGHT_BASE,BETA_BASE)
> 56
> 57
> 
> in YOLO__Init_EX(In_Address, Out_Address, Weight_offset, Beta_offset, InFM_num, OutFM_num, Kernel_size, Kernel_stride, Input_w, Input_h, Padding, IsNL, IsBN, TM, TN, TR, TC, mLoops, nLoops, rLoops, cLoops, LayerType, InputQ, OutputQ, WeightQ, BetaQ, WEIGHT_BASE, BETA_BASE)
> 53
> 54 while True:
> ---> 55 ap_idle = (mmio.read(XYOLO2_FPGA_CTRL_BUS_ADDR_AP_CTRL)>>2)&0x01
> 56 if(ap_idle):
> 57 break
> 
> /usr/local/lib/python3.6/dist-packages/pynq/mmio.py in read(self, offset, length)
> 137
> 138 self._debug('Reading {0} bytes from offset {1:x}',
> --> 139 length, offset)
> 140
> 141 # Read data out
> 
> # KeyboardInterrupt:
> the last line shows keyboard interrupt which was done by me as it was taking so much time to run the code (24 min)
> 
> please help me regarding this.....
I am also modifying this code. Can I communicate with you?

chesterkyles, 2019-11-04:
You can't run two overlays at the same time. you need to incorporate both IPs in one overlay