dhm2013724/yolov2_xilinx_fpga #13: 我在PYNQ上复现您的程序，有些问题想请教一下您

xiaozhupeiqi666, 2019-03-22:
我把你的程序移植到我的zcu104上生成了bit和tcl脚本，之后根据你的步骤在jupyter上调用yolov2.ipynb文件的时候发现，调用的图片路径ORIG_IMG_PATH = 'dog.jpg' 还有bit overlay = Overlay("yolov2.bit") 以及两个权重的路径好像都是用的你自己的路径，在我这边执行的话就会报错找不到相关文件，所以我想问一下你像bit、coco.names、dog.jpg、weightsv2_comb_reorg_ap16.bin这些文件都应该放在哪里，我都移植在了pynq目录下的overlays下仍然有很多错，所以希望你能够解答一下疑惑，我是python的小白，硬转软刚入的坑

Accioy, 2019-06-27:
您好，我也想在zcu104上复现这个项目。按照hls,vivado,pynq流程做了一遍，到jupyter里跑到最后这里卡住不动了，也没有报错信息不知道原因。请问是前面硬件部分需要做什么改动吗？
[image]

clancylea, 2019-06-27:
> 您好，我也想在zcu104上复现这个项目。按照hls,vivado,pynq流程做了一遍，到jupyter里跑到最后这里卡住不动了，也没有报错信息不知道原因。请问是前面硬件部分需要做什么改动吗？
> [image]

你好，请问你硬件重新综合了吗，需要适配zcu104板卡重新综合 ^_^

xiaozhupeiqi666, 2019-06-27:
加我wx：cyl875013745详细说说吧。

wgq18, 2022-10-18:
您好，我用zcu104 pynq复现工程遇到了问题，可以加vx咨询吗？vx：15352208338
@xiaozhupeiqi666 @clancylea

xiaozhupeiqi666, 2022-10-18:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。