dhm2013724/yolov2_xilinx_fpga #14: 有关VIVADO的BLOCK DESIGN的问题

yatingdai, 2019-04-02:
您好~我导入的PROCESSING_SYSTEM7_0跟您的相比少了很多的管脚，因为第一次接触vivado，所以不知道您教程中的add ip ps7.0 apply configuration pynq_revC.tcl是什么意思，我在tcl console中运行了这个tcl 文件但是好像也没有什么变化，所以想来向你请教一下~

wei8171023, 2020-02-04:
Hi!Have you solved this question?I have the same problem with you!

qing-2, 2022-03-28:
Hello!Have you solved this question?

TSeminent, 2023-03-05:
> 您好~我导入的PROCESSING_SYSTEM7_0跟您的相比少了很多的管脚，因为第一次接触vivado，所以不知道您教程中的add ip ps7.0 apply configuration pynq_revC.tcl是什么意思，我在tcl console中运行了这个tcl 文件但是好像也没有什么变化，所以想来向你请教一下~

您好，请问您解决这个问题了吗，我也遇到了相同的问题

TSeminent, 2023-03-05:
> Hello!Have you solved this question?

Hello! Have you solved this problem ? I have the same problem now

wyk216, 2023-06-29:
@ qing-2 ，求问大佬怎么解决的

wyk216, 2023-06-29:
@qing-2 ，求问大佬怎么解决的