dhm2013724/yolov2_xilinx_fpga #18: what is weight and beta offset ?

Thilanka97, 2019-05-15:
@dhm2013724 Hey,
I am trying to understand how your implementation works and I do not understand what weight offset and beta_offset is. Also What it does? Please explain.

Thanks in advance!

liutongxue123, 2019-05-16:
> @dhm2013724 Hey,
> I am trying to understand how your implementation works and I do not understand what weight offset and beta_offset is. Also What it does? Please explain.
> Also is inputQ, outputQ, weightQ and betaQ mean quantized input, quantized output, quantized weight and quantized beta? do you input both quantized weight, beta files and 32fp weight, beta files to the design? (I am confused because there are number of weight and beta files) what is the difference between weight and weightQ? how are they used in the design ?
> 
> Also what is Tr and Tc? and mloops, nloops,rloops and cloops?
> 
> Thanks in advance!

我也在修改此代码 方便一起交流下么

Thilanka97, 2019-05-16:
@liutongxue123 nice! do you understand what is the difference between weight and weightQ? how are they used in the design ? what  inputQ, outputQ, weightQ and betaQ  are?

Thilanka97, 2019-05-20:
@dhm2013724 
how did you select the Tn, Tm, Tr and Tc <2,32,26,26> values?

Thanks in advance!

JinzhongHe, 2019-05-22:
建议各位多看看下面引用的论文吧 很有用 现在差不多整明白这些细节上的东西是怎么实现的了

Thilanka97, 2019-05-23:
@JinzhongHe I read some. But I can not get the full pdf versions of these. I can see only the abstract.
[6] A Dynamic Multi-precision Fixed-Point Data Quantization Strategy for Convolutional Neural Network
[7] Caffeine: Towards Uniformed Representation and Acceleration for Deep Convolutional Neural Networks

can u send m if you have full versions. Also from which paper can you get an idea about inputQ, outputQ, weightQ and betaQ?

Thanks in advance!

JinzhongHe, 2019-05-28:
Do you use QQ?Maybe we can add friends for convennient communication.My QQ number is 1217283866.

Thilanka97, 2019-05-30:
@JinzhongHe Sorry I dont use QQ

Thilanka97, 2019-06-07:
@JinzhongHe hey I can not download the weight files from the baidu cloud. Can u send it to me if you have them downloaded. 
email- thiltt97@gmail.com 

Thanks in advance!