dhm2013724/yolov2_xilinx_fpga #19: vivado 2018.3 hls problem

Diezel66, 2019-06-02:
when I run hls, there was a problem:

ERROR: [HLS 200-70] Compilation errors found: In file included from cnn.cpp:1:
cnn.cpp:172:59: error: conditional expression is ambiguous; 'typename ap_int_base<6, false>::RType<3, false>::plus' (aka 'ap_uint<7>') can be converted to 'int' and vice versa
 ap_uint<6> T2R_bound = ((t2_local + T2R)<(((26 -1)*2 +3))?(t2_local + T2R):(((26 -1)*2 +3)));

how can I solve it?Thank you very much!

Diezel66, 2019-06-03:
> when I run hls, there was a problem:
>
> ERROR: [HLS 200-70] Compilation errors found: In file included from cnn.cpp:1:
> cnn.cpp:172:59: error: conditional expression is ambiguous; 'typename [...]

closed 2019-06-03 by Diezel66

Accioy, 2019-06-18:
I met the same problem. Have you solved it?

Diezel66, 2019-06-18:
it very simple,150MHz clock brank  seems to have bug. you should choose another branch and run it