dhm2013724/yolov2_xilinx_fpga #21: 生成ip时候的warning信息

njuhang, 2019-06-04:
运行hls下的script.tcl出现了很多warning，包括第一行出现的"Unknown Tcl command ' vivado_hls -f script.tcl' sending command to the OSshell for execution. It is recommended to use 'exec' to send the command to the OS shell"，以及后面好像跟具体设计有关的warning，可以直接忽视吗？

clancylea, 2019-06-04:
你好，请问你用的是vivado HLS 2018.2 版本吗，你可以自己用HLS创建工程，添加文件。

clancylea, 2019-06-04:
> 运行hls下的script.tcl出现了很多warning，包括第一行出现的"Unknown Tcl command ' vivado_hls -f script.tcl' sending command to the OSshell for execution. It is recommended to use 'exec' to send the command to the OS shell"，以及后面好像跟具体设计有关的warning，可以直接忽视吗？

  @njuhang 你好，请问你用的是vivado HLS 2018.2 版本吗，你也可以自己用HLS创建工程，添加文件。

njuhang, 2019-06-04:
> > 运行hls下的script.tcl出现了很多warning，包括第一行出现的"Unknown Tcl command ' vivado_hls -f script.tcl' sending command to the OSshell for execution. It is recommended to use 'exec' to send the command to the OS shell"，以及后面好像跟具体设计有关的warning，可以直接忽视吗？
> 
> @njuhang 你好，请问你用的是vivado HLS 2018.2 版本吗，你也可以自己用HLS创建工程，添加文件。

好的谢谢

njuhang, 2019-06-05:
> > 运行hls下的script.tcl出现了很多warning，包括第一行出现的"Unknown Tcl command ' vivado_hls -f script.tcl' sending command to the OSshell for execution. It is recommended to use 'exec' to send the command to the OS shell"，以及后面好像跟具体设计有关的warning，可以直接忽视吗？
> 
> @njuhang 你好，请问你用的是vivado HLS 2018.2 版本吗，你也可以自己用HLS创建工程，添加文件。

你好，我用vivado hls图形界面做了一遍，除了我上面打出来的那个warning没有出现外其他的输出信息包括warning都是一样的，而且在vivado中Package 此ip的时候在"Ports and Interfaces"这一栏始终是感叹号，Package结果会报很多critical warning，但是在Block Design里面添加此ip的时候引脚是没有问题的。不知道是不是可以忽视上面的问题继续下去。

Thilanka97, 2019-06-07:
@njuhang @clancylea Do you guys know why this code uses 416 x416 input size while yolov2 uses 608x608? 
Also how to change this to run on zedboard ?

Thanks in advance!

clancylea, 2019-06-09:
> @njuhang @clancylea Do you guys know why this code uses 416 x416 input size while yolov2 uses 608x608?
> Also how to change this to run on zedboard ?
> 
> Thanks in advance!

if you want to run on the zedborad , you can use  C driver and application in the vivado SDK. And you also need the at least 120M Shared Memory for FPGA. good luck!

Thilanka97, 2019-06-10:
@clancylea what do you mean by c driver ? shared memory means ? RAM ? How about the vivado design ? 
Thanks in advance!

clancylea, 2019-06-10:
> @clancylea what do you mean by c driver ? shared memory means ? RAM ? How about the vivado design ?
> Thanks in advance!

● C driver mean the fpga driver（read/write register with python ）rewrite used c language and the application used this project repo /Software version.
● Shared memory mean changing zedboard device tree ，share DDR memory 256MB for FPGA (change mem from 0x00000000 to 0x100000000).
● vivado design doesn't need to change.

Thilanka97, 2019-06-11:
@clancylea thank you so much for the reply. It helps alot. I am still new to this area. 
So the hls code doesnt need to change, you have to export the rtl design and make the vivado design(same as pynq) and then the python software code for pynq should be re-written in C language? Am I correct ?

> Shared memory mean changing zedboard device tree ，share DDR memory 256MB for FPGA (change mem from 0x00000000 to 0x100000000)

where should you do this ?

Thanks in advance!

Thilanka97, 2019-06-12:
@clancylea is the python software code the interface between pc and FPGA ?

TSeminent, 2023-03-05:
> if you want to run on the zedborad , you can use C driver and application in the vivado SDK. And you also need the at least 120M Shared Memory for FPGA. good luck!

Hello! Can you explain more detailed how to do this work with the repo SDK in this project? I have download Xilinx SDK2019.1 but I don't know how to use it with the sources in the repo SDK. Thanks in advance