dhm2013724/yolov2_xilinx_fpga #23: 复现这个工作的时候遇到的问题

CARP2079, 2019-06-12:
你好，我也是搞这方面的研究生，有幸看到文章并尝试复现这工作，然而，运行tcl文件后发现：
1、你tcl中时钟设置的是5.2ns，在你默认fpga选型上时序到不了。（150MHZ也不行）
2、各项资源运行后都与论文的不符，而且差的较多。我的软件版本与你一致，是不是有一些指令没有以source存储而是在directives.tcl里，而你没传这个文件呢，或者这个project有configuration settings，没有说明。
3、关于pingpong操作，hls自带的dataflow命令用着也挺好的，手动设置标志位是为了更好的控制还是有啥别的考虑？为什么不用官方的ap_fix来修饰变量啥的？
4、还有从你给代码中，你测试正确性是把代码放yolov2.h中在VS里面测的，也就是csim的正确性OK，但是好像没有TB，没有hls中的cosim功能。
5、这个是按照4*32展开的，我之前也写了类似的卷积核，在一些情况下有速度的制约点，我希望加一下我微信交流一下。微信号bestcarp。

dhm2013724, 2019-06-12:
兄弟，我已经要毕业了，最近会把整个project再整理一下，但是只是在Zedboard上的，Python部分不是我负责的；先回答你的问题：
（1）就是为了能到150MHz，才选的5.2ns；HLS C综合的报告说的是最坏情况，你自己算算5.2ns的时候，加上报告里面不确定的部分能不能到150M（6.6667ns）。（这个是个小诀窍，用更高的时钟综合出低时钟的设计，比较常用）
（2）我基本上跑的资源都跟论文里面差不多，我不知道你跑的是哪个代码，我觉得很奇怪。。。这个问题我回答不了你，所有的directives都是用的source（兄弟。。。你真的跑过我那个代码吗。。。我一般上传前都会从头到尾试一遍的，没问题才上传的）
（3）我咨询过很多大哥，dataflow比较脆弱，具体适用场合你可以参考UG902，一般只适用于串行流的形式；为什么不用ap_fix，因为我用的动态定点16位量化，小数点是会变的，然后正好是带符号16位，考虑到在VS里也能跑，直接上short类型就够了。
（4）正确性方面：整个网络的RTL仿真时间太久，还经常溢出，你要是愿意，可以试试分层的RTL仿真，应该是OK的；我是已经得到实际在板子上加速器跑完的结果了，跟VS软件仿真的一致，所以这边就没搞RTL仿真的部分。
（5）实际在足够大的情况下确实有制约，之前考虑过参考FPGA 2017 Dr. Ma的那篇从输出特征图宽高的维度展开，可惜我已经毕业了，凉凉~也可以考虑脉动阵列，Jason Cong教授之前好像有新文章；这方面已经比较成熟了，目前关键点已经转到软硬件协同（编译优化+架构优化协同），这个就比较难了~水个毕业跑路了~~
你这个搞的我很火冒，大兄弟；搞得跟假的一样，谁假的还开源。。。交流的话，可以拉个微信群聊聊，不过这方面我之后应该是不会继续做了，毕竟已经毕业了。最近应该会晚上更新项目，白天要上班，以上

CARP2079, 2019-06-13:
别火冒啊（火冒这词，无锡人？），我本意也只是真的遇到了这些问题想交流和询问一下，毕竟搞这个的和计算机比太少，加个微信交个朋友吧。

closed 2019-06-14 by CARP2079