dhm2013724/yolov2_xilinx_fpga #24: what are beta_offset and weight offset and INTER_WIDTH (19)?

Thilanka97, 2019-06-24:
Hey,
Could anyone please explain or tell me what beta_offset and weight offset and INTER_WIDTH (19) mean in this project ? It will be a great help. 
Thanks in advance!

dhm2013724, 2019-06-24:
Acutally, beta_offset and weight_offset are the address offset for different layers.  INTER_WIDTH is a constant value that similar with InputQ or OutputQ to indicate the fraction bit.

Thilanka97, 2019-06-24:
@dhm2013724 Thank u for the reply. 
how do u find the beta and weight offset values ? Also do u think its possible to change this to use  INT8 weights and biases and int8 operations?
Also what are *Memory_bottom, *Memory_top and *Memory_buf ? whats the difference between these?
It will be a great help!
Thnaks in advanve!

dhm2013724, 2019-06-24:
There are nearly 30 layers in YOLOv2 model. For convolutional layer, it needs input/output feature maps and kernel weights. In software version, I just used two continued buffers to store feature maps and weights for simulating memory access operations. Like stack and heap in software, I manually set the push/pop address in two buffers for different layers.
Memory_buf means the memory buffer for feature map, and  Memory_bottom & Memory_top are the upper and lower limitation for stack/heap.
If you want to use INT8 percison, you should manually concat 8bit data to 32/64 or other datawidth for data transmission between ip and memory.

Thilanka97, 2019-06-24:
@dhm2013724 thank u sooo much for the reply. 

> If you want to use INT8 percison, you should manually concat 8bit data to 32/64 or other datawidth for data transmission between ip and memory.

is that the method you use now for 16bit weight ? where do I have to change ? cnn.cpp and yolov2.h right ? Do I have to change the vivado design ? 

Thanks in advance!

Thilanka97, 2019-06-24:
@dhm2013724 
_cnn/cnn/solution1/sim/verilog/AESL_axi_master_DATA_BUS.v: Read request address       1040 exceed AXI master DATA_BUS array depth:       1024
$finish called at time : 5265 ns : File "/home/thilanka/Documents/Documents/github/fpga/vivado_tests/cnn/cnn/solution1/sim/verilog/AESL_axi_master_DATA_BUS.v" Line 700
 quit_

I am getting this error in the rtl co simulation. what can be wrong here ? 
Also this is the output I am getting in the c simulation. I do not understand why I am getting like this. 
please guide me. 

[image]

Thanks in advance!

dhm2013724, 2019-06-24:
It depends on the actual design. For example, if you want to use INT8 percision and the datawidth of the data transmission interface is 64bit, you have to wait for 8 continued datas and concat these 8 datas to one 64-bit data for transmission. In accelerator design, you should modify the input\output module. If further optimizing currecnt design, I suggest to use AXIStream interface with SG DMA.

dhm2013724, 2019-06-24:
> @dhm2013724
> _cnn/cnn/solution1/sim/verilog/AESL_axi_master_DATA_BUS.v: Read request address 1040 exceed AXI master DATA_BUS array depth: 1024 $finish called at time : 5265 ns : File [...]

HLS has told you, the depth of AXI master is too small, you should use bigger depth. For RTL-cosimulation, I suggest simulation single layer for sparing time.  If C-simulation is ok and set the big enough depth, maybe the accelerator structure has somethings wrong, you should look at the wave to find the bug module.

Thilanka97, 2019-06-24:
@dhm2013724 C cimulation doesnt give any errors but gives wrong results as shown in the image I posted. It doesnt predict correctly. Why is that ? what can be wrong ? 

> HLS has told you, the depth of AXI master is too small, you should use bigger depth

This is the AXI depth right ? This is set to 512 by you in the directive. Why do you thinkit does not work with mine. I didnt change anything. 

Thanks in advance!

Thilanka97, 2019-06-24:
> In software version, I just used two continued buffers to store feature maps and weights for simulating memory access operations
@dhm2013724 @clancylea 
are main.cpp and yolov2.h files testbench files for the cnn.cpp file for vivado_hls simulation. because without testbench files I can not run cnn.cpp file in xilinx vivado_hls. so I set cnn.cpp as the main file and main.cpp, yolov2.h and weight files as test bench files in the vivado_hls project. Am i doing it correctly ? Am I doing anything wrong? please guide me.

Thanks in advance!

Thilanka97, 2019-06-25:
@dhm2013724 what is the difference between cnn.cpp(hardware code) and yolov2.h, main.cpp(software code)? because these files contain the same functions and codes. Please guide me. Also is there a way to contact you other than github. maybe email?
Thanks in advance! 

Thank you.

Thilanka97, 2019-06-25:
@dhm2013724 what are the improvements you have made from your first version(n=2, m=32) to the current version. 
I see that you have added some parts with burst lengths and have changed the weight and input load parts. Whats the difference between anti-reorg and reorg version. What is weight reorg means?

 Also what do u mean by change the hardware and generate new weights? Does that mean that the weights change with the hardware ? 

Also if you want to use this for tiny yolov2 how to use the weights ? can you provide the files for tiny yolov2 ? 

Thanks in advance!