dhm2013724/yolov2_xilinx_fpga #26: 微信群

dhm2013724, 2019-06-28:
创了一个微信群，现阶段较忙，更新会比较慢，有问题可以先加群大家讨论

Thilanka97, 2019-06-28:
@dhm2013724 I am facing difficulty to log in to vchat. I dont know why. There is verification issue with security. Anyways I will try to fix it and connect. Could you please share the files needed to run on zedboard? my email - thiltt97@gmail.com

Thanks in advance!

ruibao123, 2019-07-08:
您好，想要加群参与讨论，能再分享一下二维码吗？万分感谢！！！

clancylea, 2019-07-08:
> 您好，想要加群参与讨论，能再分享一下二维码吗？万分感谢！！！

[image]

linbaiwpi, 2019-07-26:
二维码过期了，可否再分享下？

zhouzj01, 2019-09-02:
您好，像加群讨论，可是二维码过期了，能再次分享一下吗？万分感谢！

clancylea, 2019-09-02:
[image]

clancylea, 2019-09-02:
[image]

clancylea, 2019-09-02:
> 您好，像加群讨论，可是二维码过期了，能再次分享一下吗？万分感谢！

[image]

budaidai, 2019-09-15:
您好，可以再分享一下微信群吗？

clancylea, 2019-09-15:
> 您好，可以再分享一下微信群吗？

[image]

budaidai, 2019-09-15:
> > 您好，可以再分享一下微信群吗？
> 
> [image]

谢谢

DicerZHQ, 2019-09-29:
> > > 您好，可以再分享一下微信群吗？
> > 
> > 
> > [image]
> 
> 谢谢
可以再分享下微信群不？2019.9.29现在已经过期了哈

DicerZHQ, 2019-09-29:
> 二维码过期了，可否再分享下？

你好，可以再分享下微信群吗？2019.9.30已经过期了哈

clancylea, 2019-09-29:
> > 二维码过期了，可否再分享下？
> 
> 你好，可以再分享下微信群吗？2019.9.30已经过期了哈

[image]

DicerZHQ, 2019-09-29:
> > > 二维码过期了，可否再分享下？
> > 
> > 
> > 你好，可以再分享下微信群吗？2019.9.30已经过期了哈
> 
> [image]

谢谢哈！

DirectionLynn, 2019-10-09:
哪位还有二维码的能不能更新下微信群连接

DicerZHQ, 2019-10-09:
> 哪位还有二维码的能不能更新下微信群连接

[image]

baipengcheng, 2019-10-22:
哪位还有二维码的能不能更新下微信群连接,谢谢！

clancylea, 2019-10-23:
> 哪位还有二维码的能不能更新下微信群连接,谢谢！

[image]

baipengcheng, 2019-10-25:
> > 哪位还有二维码的能不能更新下微信群连接,谢谢！
> 
> [image]

谢啦

goldwind950306, 2019-11-04:
哪位大佬再分享一下二维吗

xiaoguoer, 2019-11-06:
哪位大佬再分享一下二维码

baipengcheng, 2019-11-07:
> 哪位大佬再分享一下二维吗

[image]

txhan, 2019-11-22:
哪位大佬再分享一下二维码

wei8171023, 2020-02-04:
哪位大佬再分享一下二维码

yuanyuehen, 2020-02-20:
可以分享一下二维码吗 谢谢

wei8171023, 2020-02-20:
> 可以分享一下二维码吗 谢谢
你加群 了啊？我微信cui2014911，一起交流啊

baipengcheng, 2020-02-20:
可以分享一下&#x4E8C;维码吗 谢谢

  
&mdash;

wei8171023, 2020-02-20:
您好！看到了您的回复，请问您在那个《真FPGA相关群》微信群吗？可以分享一下二维码吗？谢谢了！我的微信cui2014911

 
 
   
 可以分享一下&amp;#x4E8C;维码吗 谢谢 

    
 &amp;mdash;

wei8171023, 2020-02-20:
您好！可以分享一下那个微信群二维码吗

timetraveler223, 2020-03-16:
大大 可以再分享一下那个微信群么

clancylea, 2020-03-17:
> 大大 可以再分享一下那个微信群么

[image]

LEE312-ZJU, 2020-03-18:
您好~最近正在学习您的这份代码，有诸多不懂的地方想要请教，可以再分享一下微信群的二维码吗？我的微信是18279462878~万分感谢！

RosabanksiaeAit, 2020-03-30:
求再分享一下微信群，谢谢！

zy981126, 2020-04-05:
能再分享下微信群的二维码吗 非常感谢！

dhm2013724, 2020-04-06:
[image]

albb287604748, 2020-05-09:
老师您好，我这想学习您的工程，请问能加下微信群吗？之前的失效了

timetraveler223, 2020-05-09:
我不是老师。。。。 我也是想向大佬学习的本科生= =  你加我微信吧 我拉你入群wxid_280vw894knox22

albb287604748, 2020-05-09:
>  我不是老师。。。。 我也是想向大佬学习的本科生= =&amp;nbsp; 你加我微信吧 我拉你入群wxid_280vw894knox22
>
>
>
>
>
>
>  ------------------&amp;nbsp;原始邮件&amp;nbsp;------------------

十分感谢您，但我微信没搜到，您方便加我吗？ 我的微信wangmeng9324

mengbozhang2020, 2020-06-01:
> 十分感谢您，但我微信没搜到，您方便加我吗？我的微信wangmeng9324
> […](#)
> ------------------＆nbsp;原始邮件＆nbsp; ------------------发件人：“ timetraveler223 “ <notifications@github.com> ;; 发送时间：＆nbsp; 2020年5月9日（星期六）晚上9:50收件人：“ dhm2013724 / yolov2_xilinx_fpga” <yolov2_xilinx_fpga@noreply.github.com> 抄送：“王蒙” <287604748@qq.com> ;;“评论” <comment@noreply.github.com> ;; 主题：＆nbsp; Re：[dhm2013724 / yolov2_xilinx_fpga]微信群（[＃26](https://github.com/dhm2013724/yolov2_xilinx_fpga/issues/26)）我不是老师。。。。我也是想向大佬学习的本科生= =＆amp; nbsp; 你加我微信吧我拉你入群wxid_280vw894knox22
> ------------------＆amp; nbsp;原始邮件＆-nbsp; ------------------发件人： “ albb287604748” <notifications@github.com& gt ;; 发送时间：2020年5月9日（星期六）晚上9:46收件人：“ dhm2013724 / yolov2_xilinx_fpga” <yolov2_xilinx_fpga@noreply.github.com& gt ;; 抄送：“李骁” <1416974349@qq.com& gt; “评论” <comment@noreply.github.com& gt ;; 主题：Re：[dhm2013724 / yolov2_xilinx_fpga]微信群（[＃26](https://github.com/dhm2013724/yolov2_xilinx_fpga/issues/26)）老师您好，我这想学习您的工程，请问能加下微信群吗？直接回复此电子邮件，在GitHub上查看或取消订阅。—您收到此评论是因为您发表了评论。直接回复此电子邮件，在GitHub上查看或取消订阅。
请问您加群了吗。我加你微信了，可以拉我一下吗？

Rubick-fafafa, 2020-06-19:
> 我不是老师。。。。 我也是想向大佬学习的本科生= =  你加我微信吧 我拉你入群wxid_280vw894knox22
> […](#)

你好 可以拉我进群吗 我的微信wxid_e6r9a3zq33h522

xbdxwyh, 2020-07-09:
您好，方便再分享一下二维码或者麻烦您拉我进群吗？我的微信：wyh199808280350，非常感谢。

muzixilin, 2020-07-12:
您好，同楼上，同求二维码或者麻烦好心人拉我进群，我的微信：lixilin0603，非常感谢。

kangyuzhe666, 2020-07-13:
在复现您的工程过程中遇到一些问题，放不方便拉我进群，我的微信：13777453014，邮箱：1399109998@qq.com

BCNK, 2021-05-18:
在将网络映射到zedboard上遇到一些困难，方便发一下二维码 或者拉一下群吗，vx：xxfbcnk

gexianwei1231, 2021-05-22:
可以再分享一下群二维码吗？或者有哪位好心人拉我一下，我的微信是17806227597

gexianwei1231, 2021-05-22:
可以再分享一下群二维码吗？或者有哪位好心人拉我一下，我的微信是GalaxyXW1998

cwandzp, 2022-12-19:
你还，能再分享一下微信群二维码吗，或者方便拉我进微信群吗，我的微信是：lhj1036822390

Ajfil, 2023-03-14:
您好，可以请大佬拉我进微信群吗，我的微信是hbc15527345779

Kangoing, 2024-12-12:
您好，想一起讨论交流，有好心人可以拉我进一下微信群吗？我的微信是KYMNJ23

llyyf, 2025-03-27:
您好，有大佬可以拉我进下群吗？我的微信是llyyyyffff

baipengcheng, 2025-03-27:
群二维码，不过很少人讨论

白鹏程
***@***.***