dhm2013724/yolov2_xilinx_fpga #27: What does the "T2R" in the input load stand for ?

ms0488638, 2019-07-08:
Hi, I've noticed there is a variable named T2R existed in input load, and this variable  is only asserted as 4, 2 and 1 based on the input Tile width.

However, this value is then send in to another sub-function "copy_input2buf_row."

In this sub-function, T2r will then change the loop constraint.

So, what's the purpose of this variable, what does it stand for?

Thanks!

dhm2013724, 2019-07-08:
The parameters like Tn, Tr, Tc, Padding, N, R, and C will decide the load phase of input module. Because FPGA's on-chip storage is too small, we have to load some tiles each time. And the tiles' height and width will also affect actual bandwidth. Considering the input line buffer size is too big for some small scale feature maps, the input module will load several continous rows pixels each time. 
The variable T2R is used to decide the number of continous rows that be read each time. For example, if input width is 26, current input buffer can read continous 26*2 pixels each time; if input width is 13, it can read continous 13*4 pixels each time. But for other bigger width, it can only read one line each time.

ms0488638, 2019-07-08:
So Is the pixel stands for the real “pixel,” or the Input that you glued two pixel together?

dhm2013724, 2019-07-08:
> So Is the pixel stands for the real “pixel,” or the Input that you glued two pixel together?

Here, the pixel is related to real data transmission, and it should be int32 that concat two fixd16 pixel values.

ms0488638, 2019-07-08:
> For example, if input width is 26, current input buffer can read continous 26x2 pixels each time; if input width is 13, it can read continous 13x4 pixels each time. But for other bigger width, it can only read one line each time.

One more thing, how do you decide these parameters? Are they based on the burth length of AXI or something else?

Thanks!

dhm2013724, 2019-07-08:
> One more thing, how do you decide these parameters? Are they based on the burth length of AXI or something else?
>
> Thanks!

The hardware parameters like Tm, Tn, Tr, Tc decide the hardware storage and parallelism.  For each layer, the related input variable TM, TN, TR, TC should be not bigger than these hardware parameters. And for the data transmission,  considering that AXI burst length varies from 1-256, the larger burst length corresponds to the larger actual bandwidth. So, if the input/output feature maps are continous, we consider using big enough burst length to transmit them.

ms0488638, 2019-07-08:
Thank you, it's really helpful.

closed 2019-07-08 by ms0488638