dhm2013724/yolov2_xilinx_fpga #28: What's the purpose for Implementing 4 Input?

ms0488638, 2019-07-25:
Hi, I've noticed your HLS code contain 4 same Input addresses, and they were sent to mmcpy_inputport0~3 respectively. 

I wonder what is the purpose of doing this?

Can't we only use 1 input address, then send them to 4 distinct mmcpy_inputport?

Thanks.

dhm2013724, 2019-07-26:
> Hi, I've noticed your HLS code contain 4 same Input addresses, and they were sent to mmcpy_inputport0~3 respectively.
> 
> I wonder what is the purpose of doing this?
> 
> Can't we only use 1 input address, then send them to 4 distinct mmcpy_inputport?
> 
> Thanks.

Yes, off course. You can also use one channel to transmit multilpe input feature maps serially. Here, because HP ports' bandwidth is big enough, I use 4 input channels to transmit datas concurrently.

ms0488638, 2019-07-26:
> Here, because HP ports' bandwidth is big enough

Is there any way to evaluate such issue?

Ahhhhh, quite Few data are released from Xilinx to model the input channels and HP port.