dhm2013724/yolov2_xilinx_fpga #30: The allocated Memory Size in Python Code

ms0488638, 2019-08-07:
Hi, I'm a little bit confused with the python code.

In which, you allocated the memory as follow
```
img_base_buffer = xlnk.cma_array(shape=(4194304,), dtype=np.int32)
print("16M",img_base_buffer.physical_address)
IMG_MEM = img_base_buffer.physical_address
```
However, I am not quite sure about how does "4194304" come up.

I mean, in the following code, you declared some length like this

```
MEM_BASE = IMG_MEM
MEM_LEN = 416*416*32*2+208*208*32*2
Memory_top = MEM_BASE
Memory_bottom = MEM_BASE + MEM_LEN
```
I guess the MEM_LEN stand for the maximum data you have to keep (Input and output of  layer1).

So, it's more like "416*416*32* 2(Byte)++208*208*32*2(Byte)=3461120 (int 32 Word)"

However, this number still far from the "4194304"?

Did I miss anything?

Thx!!

dhm2013724, 2019-08-12:
The python code was written by following the C++ codes in SDK. And, I just maunally set every input/output feature maps' base address by simualting the input and output process in one buffer like heap and stack.

ms0488638, 2019-08-12:
> And, I just maunally set every input/output feature maps' base address by simualting the input and output process in one buffer like heap and stack.

Yes, I am curious how do yo decide the size of the buffer.

The size is different between Python and C++ SDK.

dhm2013724, 2019-08-13:
Really? Oh, I'm not familar with the python version. But in SDK version, just follow the forward network direction, I considered every layer's input and output, put input in the top of the buffer, and put output in the bottom of the buffer. When it turns to next layer, just in turn.  In addition, if it comes to route layer, multi-layers' input and output should be considered.

dhm2013724, 2019-08-13:
And then, find the maximum buffer size of the whole network, this is the final buffer size of memory buffer.

closed 2019-08-24 by ms0488638