dhm2013724/yolov2_xilinx_fpga #32: 使用vivado SDK按照作者的petalinux生成的镜像，无法读取yolov2.cfg文件

ning11111, 2019-09-05:
使用vivado SDK按照作者的petalinux生成的镜像，无法读取yolov2.cfg文件，希望大佬给与解答，感激不尽，
[image]

dhm2013724, 2019-09-05:
兄弟，你visual studio版报错吗？那个不报错，这个应该也不会啊。。。

ning11111, 2019-09-05:
HLS仿真的跑过了 然后用官方给的pynq2.3版本镜像跑pynq代码也跑通了，当我尝试去使用SDK的时候，用petalinux自己生成镜像就无法读取yolo.cfg文件，用官方的镜像事可以读取yolo.cfg文件的，希望大佬给解答一下，感激不尽

ning11111, 2019-09-05:
HLS仿真的跑过了 然后用官方给的pynq2.3版本镜像跑pynq代码也跑通了，当我尝试去使用SDK的时候，用petalinux自己生成镜像就无法读取yolo.cfg文件，用官方的镜像是可以读取yolo.cfg文件的，希望大佬给解答一下，感激不尽

ning11111, 2019-09-05:
vs没有报错 软件代码没有问题，可能镜像的问题，我用的是vivado 2019.1

dhm2013724, 2019-09-05:
你确定一下SDK里面用了静态链接没有？yolov2.cfg一般没问题

ning11111, 2019-09-05:
[image]
是这个吗？用了

dhm2013724, 2019-09-05:
[image]
你这说明sections->front是Null啊，

ning11111, 2019-09-05:
[image]
cfg在同一级目录下，确实是有的，而且我用官方pynq2.3去执行的时候，这一步是可以通过的，同时如果没有cfg文件，这里是要报错找不到的
[image]

ning11111, 2019-09-05:
[image]
[image]
可以发现print的n的值是1621784，并不为零，所以下一步无法通过

ning11111, 2019-09-05:
[image]
[image]
可以发现print的n的值是1621784，并不为0，所以后面无法通过

ning11111, 2019-09-05:
但是后面无法通过

dhm2013724, 2019-09-05:
> 但是后面无法通过

你可以试试，用同样的cfg在vs里试试，然后比对一下read_cfg里的各部分的输出

ning11111, 2019-09-05:
[image]
[image]
[image]
对不起，我刚才表述有误，应该是可以看到第一个打印数值file是1621784，后面第二个打印数值n值为0，说明文件读入但是输出是0，应该是中间这一小段代码出现了问题

ning11111, 2019-09-05:
感谢您的指点，发生了一个很奇怪的现象就是这个cfg文件点进去看起来和正常的一样，但是文件大小却是0，可能实我在复制的过程中出现了问题，我有重新替换一个以后，代码即可运行，感谢指点，

closed 2019-11-16 by dhm2013724