dhm2013724/yolov2_xilinx_fpga #34: how to import the ip to vivado?

overing1314, 2019-09-25:
Since I want to recreate the bitstream file, I cannot import the yolov2_FPGA block diagram by followed the README.m and only found the PYNQ-Z1_C.xdc and pynq_revC.tcl.
1. import ip from ..\hls\yolov2_ap16_n2m32_inburst\solution1\impl\ip and add ip Yolo2_fpga
2. add ip ps7.0 apply configuration pynq_revC.tcl
   add constraint PYNQ-Z1_C.xdc

How I going to import the yolov2 block design by using these 2 file ?
Thank you.

dhm2013724, 2019-09-26:
It depends on your chip that you selected. The tcl and xdc files are not important, and they are just used for constructing Linux operating system. Just follow the block design, enable several HP ports to connect the ip with PS.

overing1314, 2019-09-26:
but I didn't egt any ip call "Yolo2_FPGA" 
can you share the ip file to me? or teach me how to import the ip Yolo2_FPGA.
I can not finish the block design wihout the Yolo2_FPGA ip.

dhm2013724, 2019-09-27:
The name may be changed, related codes are included in HLS dir. This doesn't matter.

SoumyaDeshapnde, 2019-10-11:
To create the ip for Yolov2..
do I have to select option 1: package current project?

[image]

dhm2013724, 2019-10-11:
> To create the ip for Yolov2..
> do I have to select option 1: package current project?
> or create an AXI4 peripheral and then add sources to it?
> 
> [image]

The YOLOv2 IP is designed by Vivado HLS, and there's a button to export the design to ip format. You  only need to import the ip that HLS exported in related Vivado project.

SoumyaDeshapnde, 2019-10-11:
> The YOLOv2 IP is designed by Vivado HLS, and there's a button to export the design to ip format. You only need to import the ip that HLS exported in related Vivado project.

okayy..
So, just to clarify, I should do a software simulation ie step 2 (HLS Accelerator and Simulation) along with the weights generated from step 1 and then find the button that exports the design to ip format, is this right?