dhm2013724/yolov2_xilinx_fpga #37: Performance Evaluation for PYNQ

chesterkyles, 2019-10-29:
Hi, I would like to ask how were you able to evaluate the performance for PYNQ in terms of Power, GOP, GOP/s? Did you use any tools?

dhm2013724, 2019-10-30:
> Hi, I would like to ask how were you able to evaluate the performance for PYNQ in terms of Power, GOP, GOP/s? Did you use any tools?

For energy consumption, I just used an power consumption tester to evaluate the board's total power. And the GOP, it sums the number of addtion and multiplication operations for every layer. To get GOP/s, just use GOP divided by latency.

chesterkyles, 2019-11-02:
> For energy consumption, I just used an power consumption tester to evaluate the board's total power. And the GOP, it sums the number of addtion and multiplication operations for every layer. To get [...]

Thank you. Sorry, I am still learning FPGA.
Follow up question, where does multiplication of image and weights happen during "yolo_fpga"? Is it during the function "YOLO__Init_EX" where writing data to MMIO takes place? Can you enlighten me with your code? Thank you.

chesterkyles, 2019-11-02:
Hi again, I've already found answers to my questions from reading documentations. The code i'd been looking for is the "cnn.cpp". Now, the only thing I need to do is add Video IP block for hdmi output. Thank you. I'll close this one.

closed 2019-11-02 by chesterkyles