dhm2013724/yolov2_xilinx_fpga #38: 大佬，您好，我想问一下测试视频的时候FPS是多少？

wpyueyue, 2019-11-04:
大佬，您好，我想问一下测试视频的时候FPS是多少？

chesterkyles, 2019-11-04:
Probably ~0.5 fps since it took 2 seconds to predict for a single image. 

[image]

Btw, how do you intend to produce the video output in a Jupyter notebook (IPython)?

dhm2013724, 2019-11-05:
我一般只管在zedboard等非PYNQ平台上的事情，PYNQ平台上的事情你得 @clancylea ，我只测过原版coco数据集的YOLOv2在Zedboard上不考虑前处理和后处理的话，差不多1秒1帧，PYNQ那个PYTHON的本来就慢，没救的。前处理的时间要看输入视频帧的大小，因为他是resize到416*416，所以肯定像素越高，时间越长；后处理主要得看能框到多少东西，框的东西越多，那肯定scan和画框的时间就越长，不确定

dhm2013724, 2019-11-05:
> Probably ~0.5 fps since it took 2 seconds to predict for a single image.
> 
> [image]
> 
> Btw, how do you intend to produce the video output in a Jupyter notebook (IPython)?

You should @clancylea for PYNQ framework