dhm2013724/yolov2_xilinx_fpga #39: PYNQ - predicting only for one class

chesterkyles, 2019-11-04:
I have a modified version of the yolov2.cfg and yolov2.weights to predict only for one class. I know for sure that I need to modify some codes especially here: 
[image]

Can you tell me what these variables are? and also how to modify in accordance to the cfg file? Can this be also applied if i want to change it to yolov2-tiny? 

Thank you. Hoping for your feedback.

dhm2013724, 2019-11-05:
> I have a modified version of the yolov2.cfg and yolov2.weights to predict only for one class. I know for sure that I need to modify some codes especially here:
> [image]
>
> Can you tell me what these [...]

You should first read some papers about quantization(references [3] and [6]), and read the software version about weight quantization.

chesterkyles, 2019-11-05:
Thank you.
I already discovered what are the variables except for inputQ, weightQ, and biasQ. If this is related to quantization then i'll be reading some papers.

chesterkyles, 2019-11-05:
Can you at least described what are these variables (inputQ, weightQ, betaQ)?

dhm2013724, 2019-11-05:
WeightQ and BetaQ are quantized params about Weight quantizations, and InputQ is about input/output feature maps quantizations. The ieda for these two quantizaiton is simliar, just to use fixed point operaitons replace float point ops. So, it needs to decide the fraction bits for every fixed point data, and it is also related to the quantization error measure after data quantization.

chesterkyles, 2019-11-07:
Thank you. Follow up question, do I still need to modify the YOLO_FPGA IP block (cnn.cpp) if i need to change it to tiny-yolov2? or just the network data? Thank you.

closed 2019-11-17 by chesterkyles