dhm2013724/yolov2_xilinx_fpga #40: LUT overflow

xiaoguoer, 2019-11-06:
Hi, I tried to run c synthesis with your code in /hls/src. And I set YOLO2_FPGA as top function.
But the result implies that LUT has overflow. And clock cannot reach the requirement either.
[image]
what's going wrong?

chesterkyles, 2019-11-07:
The author mentioned it in the Readme stating:

> Oh, this part is too complicated to brightly intoduce. Current design didnt implemment C/RTL simulation, because that testbench always overflow! if anyone can solve it, please tell me and upload it, Thanks!

dhm2013724, 2019-11-07:
> Hi, I tried to run c synthesis with your code in /hls/src. And I set YOLO2_FPGA as top function.
> But the result implies that LUT has overflow. And clock cannot reach the requirement either.
> [image]
> what's going wrong?

The hls C synthesis's resource consumption is not accurate, especially in LUT and FF. If you want to evaluate the accurate resource consumption, you can click vivado synthsis, place and route, when you want to package this ip.

closed 2019-11-16 by dhm2013724