dhm2013724/yolov2_xilinx_fpga #41: Generate bitstream error

chesterkyles, 2019-11-11:
Hi, i am trying to replicate your vivado block diagram. I just followed what is on the tcl file. However, i have timing issues during implementation.

[image]

Have you encountered these warnings? I tried to ignore them but bitstream seems not fine when i tried to use it for the overlay. Have you added any timing constraints? 

Thank you. Hoping for your feedback.

dhm2013724, 2019-11-14:
> Hi, i am trying to replicate your vivado block diagram. I just followed what is on the tcl file. However, i have timing issues during implementation.
>
> [image]
>
> Have you encountered these warnings? I [...]

I have just returned from a business trip. Yes, the timing warning can be ignored, just test it in real chip.

chesterkyles, 2019-11-15:
Okay. Thank you. I had replicated your code. I also added Video IP block and GPIO IP for the buttons, switches, and LEDs. It is running at 0.35 fps. I'm currently trying to modify it for Tiny-Yolov2 in order to achieve higher fps.

closed 2019-11-16 by dhm2013724