dhm2013724/yolov2_xilinx_fpga #43: 您好，我想问一下最终在SD卡上生成系统的问题

xbdxwyh, 2019-11-16:
[image]
我在最后生成的时候遇到了这个问题，在重新利用petalinux编译后还是这样，我想问下这个问题您遇到过吗？
还有一个问题就是在利用HLS进行C仿真的步骤中，这一步可以省略不进行吗？
感谢您的不吝赐教！

dhm2013724, 2019-11-16:
> [image]
> 我在最后生成的时候遇到了这个问题，在重新利用petalinux编译后还是这样，我想问下这个问题您遇到过吗？
> 还有一个问题就是在利用HLS进行C仿真的步骤中，这一步可以省略不进行吗？
> 感谢您的不吝赐教！

我建议你再往上找找，如果就是当前这个错误的话，你确定下uboot的config和总的config，这两者关于根文件系统的配置都没问题的话，那么就试试重新做一下文件系统在的那个区，基本没啥问题。如果是出现booti的问题就百度一下就OK；不过，我没试过2017.4，你先试试前面两个，还不成的话，先去xilinx forum上看看你这个问题找不找的到，也可能是版本bug

xbdxwyh, 2019-11-18:
真是太感谢了，根据您的提示在上面翻阅错误，发现最开头有这样的几行错误信息：
Net: ZYNQ GEM: e000b000, phyaddr ffffffff, interface rgmii-id   
PHY is not detected   
GEM PHY init failed    
No ethernet found.
我发现是因为在blockdesign中Ethernet的设置太大意的，我忘记把默认的LVCMOS和slow修改成本该是HSTL1.8V和fast的配置，现在已经解决了，感谢您的解答！

closed 2019-11-19 by xbdxwyh