dhm2013724/yolov2_xilinx_fpga #44: Can you help with Tiny Yolov2 Software Simulation?

chesterkyles, 2019-11-18:
Hi, I have modified the software simulation codes for yolov2 especially the weight_offset and beta_offset. I wanted to get the ap16 bin files for weights and beta. This is the modification:

`void yolov2_hls_ps(network *net, float *input)
{
	int x;

	network orig = *net;
	net->input = input;

	int weight_offset[16] = {432, 4608, 18432, 73728, 294912, 1179648, 4718592, 4718592, 1958400, 0, 0, 0, 0, 0, 0, 0};
	int beta_offset[16] = {16, 32, 64, 128, 256, 512, 1024, 512, 425, 0, 0, 0, 0, 0, 0, 0};
	int offset_index = 0;

	float *Weight_buf = (float *)calloc(51869376/4,sizeof(float));
	float *Beta_buf   = (float *)calloc(11876/4,sizeof(float));

#ifdef REORG_TEST
	FILE *fp_w = fopen("weightsv2_comb_reorg.bin", "rb");
    if(!fp_w) file_error("weightsv2_comb_reorg.bin");
#else
	FILE *fp_w = fopen("weightsv2_comb.bin", "rb");
    if(!fp_w) file_error("weightsv2_comb.bin");
#endif

#ifdef REORG_GEN
	float *Weight_reorg_buf = (float *)calloc(51869376/4,sizeof(float));
	FILE *fp_w_reorg = fopen("weightsv2_comb_reorg.bin", "wb");
    if(!fp_w_reorg) file_error("weightsv2_comb_reorg.bin");
#endif

	FILE *fp_b = fopen("biasv2_comb.bin", "rb");
    if(!fp_b) file_error("biasv2_comb.bin");

	fread(Weight_buf, sizeof(float), 51869376/4, fp_w);
	fread(Beta_buf, sizeof(float), 11876/4, fp_b);

	fclose(fp_w);
	fclose(fp_b);

#define MEM_LEN (416*416*16+208*208*32)
	float *Memory_buf = (float*)calloc(MEM_LEN+1024+1024,sizeof(float));
	float *Memory_top = Memory_buf+1024;
	float *Memory_bottom = Memory_top + MEM_LEN;
	memcpy(Memory_top,input,416*416*3*sizeof(float));//416x416x3 input_pic

	float* in_ptr[16];
	float* out_ptr[16];

	for(x=0;x<15;x++)
	{
		if(x%2==0)
		{
			in_ptr[x] = Memory_top;
			out_ptr[x] = Memory_bottom - net->layers[x].outputs ;
		}
		else
		{
			in_ptr[x] = out_ptr[x-1];
			out_ptr[x] = Memory_top;
		}
	}

	in_ptr[15] = out_ptr[14];

    network netp = *net;
    int i;
	int woffset = 0;
	int aoffset = 0;
	int boffset = 0;
	int TR,TC,TM,TN;
	int output_w,output_h;
	int rLoops,cLoops,mLoops,nLoops;
	double sum_gop = 0.0;

    for(i = 0; i < netp.n; ++i)
	{
        netp.index = i;
        layer l = netp.layers[i];
		printf("Layer[%2d]: ",i);
		switch(l.type)
		{
			case CONVOLUTIONAL:
				printf("outputMemory:%8d;BN=%d;Activation=%d;conv  %5d %2d x%2d /%2d  %4d x%4d x%4d   ->  %4d x%4d x%4d  %5.3f BFLOPs\n",l.outputs,l.batch_normalize,l.activation, l.n, l.size, l.size, l.stride, l.w, l.h, l.c, l.out_w, l.out_h, l.out_c, (2.0 * l.n * l.size*l.size*l.c/l.groups * l.out_h*l.out_w)/1000000000.);
				sum_gop += (2.0 * l.n * l.size*l.size*l.c/l.groups * l.out_h*l.out_w)/1000000000.;
				output_w = (l.w - l.size + 2*l.pad)/l.stride + 1 ;
				output_h = (l.h - l.size + 2*l.pad)/l.stride + 1 ;

				TR = MIN(((OnChipIB_Height-l.size)/l.stride+1),Tr);//keep Kernel_stride>=1
				TR = MIN(output_h,TR);
				TC = MIN(((OnChipIB_Width-l.size)/l.stride+1),Tc);
				TC = MIN(output_w,TC);
				TM = MIN(l.n,Tm);
				TN = MIN(l.c,Tn);

				rLoops = (int)ceil(((float)output_h)/TR);
				cLoops = (int)ceil(((float)output_w)/TC);
				mLoops = (int)ceil(((float)l.n)/TM);
			    nLoops = (int)ceil(((float)l.c)/TN);

				YOLO2_FPGA(in_ptr[i],out_ptr[i],Weight_buf+woffset,Beta_buf+boffset,
					l.c,l.n,l.size,
					l.stride,l.w,l.h,l.pad,l.activation==LEAKY?1:0,l.batch_normalize?1:0,
					TM,TN,TR,TC,
					mLoops,nLoops,rLoops,cLoops,0);
#ifdef REORG_GEN
				Weight_reorgnaization_anti(Weight_buf + woffset,Weight_reorg_buf + woffset,NULL,l.c,l.n,l.size,TM,TN,0);
#endif
				printf("TR=%d,TC=%d,TM=%d,TN=%d,rLoops=%d,cLoops=%d,mLoops=%d,nLoops=%d\n",TR,TC,TM,TN,rLoops,cLoops,mLoops,nLoops);

				woffset += weight_offset[offset_index];
				boffset += beta_offset[offset_index];
				offset_index++;

				break;
			case MAXPOOL:
				printf("outputMemory:%8d;max          %d x %d / %d  %4d x%4d x%4d   ->  %4d x%4d x%4d\n",l.outputs, l.size, l.size, l.stride, l.w, l.h, l.c, l.out_w, l.out_h, l.out_c);
				//output_w = (l.w - l.size)/l.stride + 1 ;
				//output_h = (l.h - l.size)/l.stride + 1 ;
				output_w = l.out_h;
				output_h = l.out_w;

				TR = MIN(((OnChipIB_Height-l.size)/l.stride+1),Tr);//keep Kernel_stride>=1
				TC = MIN(((OnChipIB_Width-l.size)/l.stride+1),Tc);

				TR = MIN(output_h,TR);
				TC = MIN(output_w,TC);
				TM = MIN(Tm,Tn);
				TM = MIN(l.c,TM);

				rLoops = (int)ceil(((float)output_h)/TR);
				cLoops = (int)ceil(((float)output_w)/TC);
				mLoops = (int)ceil(((float)l.c)/TM);

				YOLO2_FPGA(in_ptr[i],out_ptr[i],NULL,NULL,l.c,l.c,
					l.size,l.stride,l.w,l.h,l.pad,0,0,TM,0,TR,TC,mLoops,0,rLoops,cLoops,1);

				break;
			case REORG:
				printf("outputMemory:%8d;reorg              /%2d  %4d x%4d x%4d   ->  %4d x%4d x%4d\n",l.outputs,  l.stride, l.w, l.h, l.c, l.out_w, l.out_h, l.out_c);			
				output_w = 26;
				output_h = 32*13;

				TR = MIN(((OnChipIB_Height-l.stride)/l.stride+1),Tr);//keep Kernel_stride>=1
				TR = MIN(output_h,TR);
				TC = MIN(((OnChipIB_Width-l.stride)/l.stride+1),Tc);
				TC = MIN(output_w,TC);
				TM = 4;

				rLoops = (int)ceil(((float)output_h)/TR);
				cLoops = (int)ceil(((float)output_w)/TC);
				mLoops = 1;

				YOLO2_FPGA(in_ptr[i],out_ptr[i],NULL,NULL,1,4,
							  l.stride,l.stride,52,32*26,0,0,0,TM,0,TR,TC,mLoops,0,rLoops,cLoops,2);

				break;
			case ROUTE:
				printf("outputMemory:%8d;route ",l.outputs);
				int j;
				for(j = 0; j < l.n; ++j){
					printf(" %d", l.input_layers[j]);
				}
				printf("\n");
				break;
			case REGION:
				printf("outputMemory:%8d;Detection\n",l.outputs);
				netp.input = in_ptr[i];
				forward_region_layer(l,netp);
				break;
		}

		netp.input = l.output;

    }
	printf("SUM_GOP=%g\n",sum_gop);
	*net = orig;

#ifdef REORG_GEN
	fwrite(Weight_reorg_buf, sizeof(float), 51869376/4, fp_w_reorg);
	fclose(fp_w_reorg);
	free(Weight_reorg_buf);
#endif
	free(Memory_buf);
	free(Weight_buf);
	free(Beta_buf);
}`

However, it cannot predict. Do i still need to modify the "_YOLO2_FPGA_" function? What are the necessary modifications for the tiny yolov2? Thank you. Hoping for your reply.

chesterkyles, 2019-11-18:
Hi @dhm2013724  @clancylea , i have successfully modify it for tiny_yolov2. 
[image]

However, when running to PYNQ using the generated bin files and the values for quantization, output png has messy labels/bounding boxes.

chesterkyles, 2019-11-21:
[image]
I successfully simulated for tiny-yolov2 . However, i got problem in integrating it with PYNQ. There's probably wrong with my network data passed to the YOLO2_FPGA function.
[image]

chesterkyles, 2019-11-21:
I have found the issue. Issue is in the pooling layer.
```
output_w = (layerw[i] - layersize[i])//layerstride[i] + 1 
output_h = (layerh[i] - layersize[i])//layerstride[i] + 1  
```

closed 2019-11-21 by chesterkyles

zzulb, 2021-11-14:
Hi，@chesterkyles. Can you send the revised project file? The weight file size I used is 44906176. And the result is wrong. Thanks!

vvv241, 2023-12-26:
> output_w = (layerw[i] - layersize[i])//layerstride[i] + 1 
> output_h = (layerh[i] - layersize[i])//layerstride[i] + 1

Hi,I meet the same problem.What's wrong with the codes??