dhm2013724/yolov2_xilinx_fpga #45: 我在vivado-hls2018.2中有以下错误

wpyueyue, 2019-11-19:
您好，我在vivado-hls2018.2中复现了这个工程，c仿真和c综合都没有问题，但是在rtl仿真的时候，有以下错误。原因是AXI master 深度不够吗？
// RTL Simulation : "Inter-Transaction Progress" ["Intra-Transaction Progress"] @ "Simulation Time"
////////////////////////////////////////////////////////////////////////////////////
// RTL Simulation : 0 / 29 [n/a] @ "125000"
/home/sdsoc/yolov2-hls/yolov2/solution1/sim/verilog/AESL_axi_master_DATA_BUS3.v: Read request address     173056 exceed AXI master DATA_BUS3 array depth:        512
$finish called at time : 3285 ns : File "/home/sdsoc/yolov2-hls/yolov2/solution1/sim/verilog/AESL_axi_master_DATA_BUS3.v" Line 695
## quit
INFO: [Common 17-206] Exiting xsim at Tue Nov 19 11:50:16 2019...
ERROR: [COSIM 212-303] Aborting co-simulation: RTL simulation failed.  
ERROR: [COSIM 212-344] Rtl simulation failed.
could not read "/home/sdsoc/yolov2-hls/yolov2/solution1/sim/tv/rtldatafile/sim/report/cosim.log": no such file or directory
    while executing
"source /home/sdsoc/yolov2-hls/yolov2/solution1/cosim.tcl"
    invoked from within
"hls::main /home/sdsoc/yolov2-hls/yolov2/solution1/cosim.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$args"
    (procedure "hls_proc" line 5)
    invoked from within
"hls_proc $argv"
Finished C/RTL cosimulation.

517517517, 2020-01-09:
@wpyueyue 我遇到了和你一样的问题，请问你后来解决了吗？谢谢！

wpyueyue, 2020-01-09:
@517517517  有可能是vivado-hls c/rtl仿真的时候资源不够，作者已经提到这个问题，只要综合之后就会有ip核。所以这个问题不用管，不影响使用。

517517517, 2020-01-09:
@wpyueyue 非常感谢您！还想请问一下，我用的板子是zedboard，到了最后一步，不知道如何在SDK上运行程序，请问您成功运行过吗？非常感谢！

wpyueyue, 2020-01-09:
@517517517 我是建立的petalinux工程，在linux操作系统上跑的。

517517517, 2020-01-12:
@wpyueyue 还想请教您一个问题，请问您在最后一步petalinux用的是ubuntu16.04.几，哪个版本，以及您在ubuntu安装vivado2018.2以实现交叉编译了吗？谢谢！我的微信是wyq517517521，希望可以随时请教您，谢谢！

wpyueyue, 2020-01-12:
ubuntu16.04,，应该都可以。ubuntu下安装vivado不需要安装交叉编译环境，sdk自带交叉编译环境。按照petalinux手册搭建好工程就可以了，之后按照作者写的修改设备树。

overing1314, 2020-03-04:
@wpyueyue 可以告诉我更多吗，我最后学你成功用petalinux搭载了板子的环境了，但是问题是我该如何在板子运行yolo呢？

wpyueyue, 2020-03-04:
打开vivado sdk然后使用sdk的交叉编译。这一部分可以去csdn上找，就是简单的把程序放在sdk然后编译生成elf可执行文件，然后拷贝到板子上就可以。------------------ 原始邮件 ------------------

 
&mdash;

overing1314, 2020-03-05:
[image]
原来如此，那我想问下 在new application project 里，OS Platform 选 standalone 或着是 linux ？ Language 选 C 或是 C++ 呢？这个部分楼主没有讲述。。

wpyueyue, 2020-03-05:
linux  用c吧，我具体忘了，你可以试一下。------------------ 原始邮件 ------------------

overing1314, 2020-03-05:
@wpyueyue 我做到了elf出来，但是当我执行elf时候，他既然出现 图片乱像的问题，我从hls 到制作ip 到 vivado 连接了block diagram，最后sdk 都是跟着做。还是不明白为何会这样

overing1314, 2020-03-05:
[image]

overing1314, 2020-03-05:
@wpyueyue 我制作zedboard block diagram时候，我拿到的performance results和版住的有些出入，我的power 大概是3W ，但是他拿到1.2W，有什么因素会导致这个问题 ？我也是跟着放 100MHZ INPUT，150MHZ OUTPUT（LOW ACTIVE）

TKNopro, 2020-03-13:
我是个初学者，请问这个问题有没有比较简单的解决方法 "ERROR: [SYNCHK 200-11] ... (possible cause(s): pointer to pointer or global pointer)"谢谢

TSeminent, 2023-03-04:
> ###
> [Jan 9, 2020](#issuecomment-572480623)

也就是说作者的两个文件夹：SDK for Zedboard和Petalinux，是等价的吗？选择其中一种方法就行是吗？

TSeminent, 2023-03-04:
> ###
> [Mar 5, 2020](#issuecomment-595010204)

您好，可以私聊以下吗，我最近也在做这个项目，遇到了一些问题

TSeminent, 2023-03-05:
> @wpyueyue 我做到了elf出来，但是当我执行elf时候，他既然出现 图片乱像的问题，我从hls 到制作ip 到 vivado 连接了block diagram，最后sdk 都是跟着做。还是不明白为何会这样

您好，可以问一下您是怎么执行elf文件吗，我用Xilinx SDK编译了之后生成了elf文件但是我不清楚该怎么操作

TSeminent, 2023-03-15:
> @wpyueyue 我做到了elf出来，但是当我执行elf时候，他既然出现 图片乱像的问题，我从hls 到制作ip 到 vivado 连接了block diagram，最后sdk 都是跟着做。还是不明白为何会这样
 
您好，能问一下您用的哪个软件生成的elf吗，是在windows上操作的还是Ubuntu呢