dhm2013724/yolov2_xilinx_fpga #46: 大佬您好，感谢您的帮助，最后成功运行了

xbdxwyh, 2019-11-19:
但是这里有一个小小的问题，就是我最后的结果并不像您一样生成了prediction.png而是输出了一长串的百分百，但是也没有报错误，请问这是为什么呢？
[image]
如果要生成.png的话是需要进行修改某些文件吗？

dhm2013724, 2019-11-21:
> 但是这里有一个小小的问题，就是我最后的结果并不像您一样生成了prediction.png而是输出了一长串的百分百，但是也没有报错误，请问这是为什么呢？
> [image]
> 如果要生成.png的话是需要进行修改某些文件吗？

这我没遇到过，你先确定结果对不对，对的话你去代码里面找一找看看是不是NMS和画框那边的问题，那边我基本没改过，是darknet原来的代码；你在最后生成PNG的函数前面，先试着自己生成一张原图存下来，看看有没有；有的话，说明不是PNG生成的问题，那就去前面找；没有的话，那么估计是PNG生成函数改了，重新贴一个试试

xbdxwyh, 2019-11-22:
> > 但是这里有一个小小的问题，就是我最后的结果并不像您一样生成了prediction.png而是输出了一长串的百分百，但是也没有报错误，请问这是为什么呢？
> > [image]
> > 如果要生成.png的话是需要进行修改某些文件吗？
> 
> 这我没遇到过，你先确定结果对不对，对的话你去代码里面找一找看看是不是NMS和画框那边的问题，那边我基本没改过，是darknet原来的代码；你在最后生成PNG的函数前面，先试着自己生成一张原图存下来，看看有没有；有的话，说明不是PNG生成的问题，那就去前面找；没有的话，那么估计是PNG生成函数改了，重新贴一个试试

[image]
现在可以生成png了，不知道为什么在前面加了句printf就可以了，但是现在生成的png是这样的，把前面的if(nms)那一句注释了以后编译运行还是会生成这样的情况，不知道这样子是为什么呢

517517517, 2020-01-19:
@xbdxwyh 你好，我遇到了和你一样的问题，生成了图片不过有好多框框，请问你最后解决了吗？谢谢！

TSeminent, 2023-03-15:
> 但是这里有一个小小的问题，就是我最后的结果并不像您一样生成了prediction.png而是输出了一长串的百分百，但是也没有报错误，请问这是为什么呢？ [image] 如果要生成.png的话是需要进行修改某些文件吗？

朋友您好，可以问一下您.elf文件是怎么生成的吗