dhm2013724/yolov2_xilinx_fpga #50: 想请教您，在vivado中怎么使用pynq_revC.tcl 这个脚本

wpyueyue, 2019-11-25:
想请教您，在vivado中怎么使用pynq_revC.tcl 这个脚本。因为我自己搭建的zynq ip核出了问题，想用您的tcl脚本搭建，但是运行source pynq_revC.tcl  并没有任何反应？

dhm2013724, 2019-11-25:
> 想请教您，在vivado中怎么使用pynq_revC.tcl 这个脚本。因为我自己搭建的zynq ip核出了问题，想用您的tcl脚本搭建，但是运行source pynq_revC.tcl 并没有任何反应？

那个是基于PYNQ的，其他的板子不建议用，毕竟配置不一定一样，IP核出问题就解决IP核呗；如果你是PYNQ的配置问题，好像DDR有些地方不一样，你 @clancylea

wpyueyue, 2019-11-25:
我用的是PYNQ-z2，在vivado上搭建工程的时候，就找不到怎么修改DDR内存的地方，看到您提供了这个TCL文件。这个TCL文件好像只是修改IP核配置的，但是不知道怎么用。

wpyueyue, 2019-11-25:
[image]

clancylea, 2019-11-25:
> 我用的是PYNQ-z2，在vivado上搭建工程的时候，就找不到怎么修改DDR内存的地方，看到您提供了这个TCL文件。这个TCL文件好像只是修改IP核配置的，但是不知道怎么用。

 你好，可以添加PYNQ-Z2的boardfile，选开发板时选PYNQ-Z2的开发板，添加ZYNQ PS ip核后，run  connection automation 后DDR会自动配置为PYNQ-Z2的，这样比较方便。

Daniel-bboy, 2021-05-07:
> > 我用的是PYNQ-z2，在vivado上搭建工程的时候，就找不到怎么修改DDR内存的地方，看到您提供了这个TCL文件。这个TCL文件好像只是修改IP核配置的，但是不知道怎么用。
> 
> 你好，可以添加PYNQ-Z2的boardfile，选开发板时选PYNQ-Z2的开发板，添加ZYNQ PS ip核后，run connection automation 后DDR会自动配置为PYNQ-Z2的，这样比较方便。

您好，不好意思打扰您，因为一直问题没有得到答复，所以想问问您项目相关的问题，是否可以拉我进一下微信群，实在麻烦您了，我的微信号是845285060，谢谢！