dhm2013724/yolov2_xilinx_fpga #53: 请教下量化

xuemo188, 2019-12-09:
[image]
想问您下，为啥要乘0xccc又右移15位呢

dhm2013724, 2019-12-09:
> [image]
> 想问您下，为啥要乘0xccc又右移15位呢

因为 leaky ReLU是小于0的数乘0.1，所以这边用定点16位（1位符号位，15位小数位）来表示0.1，就是0x0ccc，你可以自己算一算

xuemo188, 2019-12-09:
谢谢您，我懂了0xccc，那为啥还要移动15位呢，换成浮点数吗

dhm2013724, 2019-12-09:
> 谢谢您，我懂了0xccc，那为啥还要移动15位呢，换成浮点数吗

这个得去看一下定点乘法，类似IEEE754，就是阶码，可以理解为乘上2^(-15)

xuemo188, 2019-12-11:
打扰您下，请教下这里为啥有个乘2呢
[image]

benpaodewali, 2019-12-12:
> 打扰您下，请教下这里为啥有个乘2呢
> [image]

请问一下这个文档是哪来的呀，想学习一下，谢谢了

dhm2013724, 2019-12-13:
> > 打扰您下，请教下这里为啥有个乘2呢
> > [image]
> 
> 应该是因为这里是双缓冲，所以要乒乓操作，乘以2。

楼上说的对，是双缓冲，所以才乘2的

xuemo188, 2019-12-14:
非常感谢两位大佬，请教下这个MM（）是什么呢
[image]

dhm2013724, 2019-12-14:
> 非常感谢两位大佬，请教下这个MM（）是什么呢
> [image]

MM论文里面有说的，应该跟内存突发读写带宽相关，你往前面的章节找一找，但是我现在觉得这边少写了几句，读写带宽应该是不一样的，我当初做的时候没考虑这点，都是按一样搞的

xuemo188, 2019-12-15:
大佬，想问您下，提供了yolo v2 hls的ip核tcl文件吗？见hls里面只有src和tb

xuemo188, 2019-12-19:
[image]
我想问上图中的Data scatter和Data gather与下面图对应不上啊
[image]

dhm2013724, 2019-12-19:
> [image]
> 我想问上图中的Data scatter和Data gather与下面图对应不上啊
> [image]

画反了= =