dhm2013724/yolov2_xilinx_fpga #54: 请教在资源更多的板子上加速运算

benpaodewali, 2019-12-10:
您好，我这儿有一块资源更多的板子，运行您的代码只用了很少的资源，现在想把资源充分利用起来，看了下您的代码，相请教一下，下面几个参数的具体含义
`#define S 2
#define K 3

#define Tn 4
#define Tm 32
#define Tr 26
#define Tc 26`
是不是修改这几个参数就可以进行加速了，谢谢了

benpaodewali, 2019-12-11:
试了一下，好像不行，看来需要往细节看了

benpaodewali, 2019-12-13:
我觉得有个地方可以改进一下，算了一下weights大小大约97MB，FM来回读写的大小只有35MB左右，可是分配4给HP接口给FM用，只有一个HP给weights，感觉有点不合理啊

dhm2013724, 2019-12-13:
> 我觉得有个地方可以改进一下，算了一下weights大小大约97MB，FM来回读写的大小只有35MB左右，可是分配4给HP接口给FM用，只有一个HP给weights，如果要加速的话，可能这儿要重新分配一下

是的，这个问题我想过；实际上，得先测下读写带宽的，最好是就4个HP，读IFM和W共享HP就可以了；带宽够也可以多加HP，或者直接接口换AXI-Stream+SG DMA；关于进一步改进，如果还是用HLS，这东西写不出很多结构，你可以先在软件版上改Tm Tn Tr Tc等等，然后等软件版那几部都对了，再移到HLS上去；软件版里幻数比较多，有很多trade off。。。现在的网络kernel都比较小了，一般就1*1和3*3，我不知道Winograd还是不是那么有效果，可能FFT会更好？或者试试Systolic Array也可以，这个HLS还是可以做的，之前看到Xilinx官方好像有example，这领域已经比较成熟了，玩玩可以，个人不建议跳坑，再往下学术贡献估计不大，Jason Cong都去搞存储了= =

benpaodewali, 2019-12-14:
> 是的，这个问题我想过；实际上，得先测下读写带宽的，最好是就4个HP，读IFM和W共享HP就可以了；带宽够也可以多加HP，或者直接接口换AXI-Stream+SG DMA；关于进一步改进，如果还是用HLS，这东西写不出很多结构，你可以先在软件版上改Tm Tn Tr Tc等等，然后等软件版那几部都对了，再移到HLS上去；软件版里幻数比较多，有很多trade [...]

非常感谢大佬回复，通过学习这份代码学习到了不少有关hls的知识，感觉比看官网教程文档收获要大很多，谢谢大佬的无私奉献。我在软件这边修改了Tm Tn Tr Tc等等，这些影响weights的排布，然后怎么修改都可以运行出正确结果，可是在fpga这边运行结果却不正确，我主要是修改Tm，把32改成128，这样乘法器就能用到512个，然后把和Tm有关的数据位宽加了两位，使用的是对应排布的weihts，这里想请教一下大佬，我这儿是不是有哪里疏漏的地方；不过确实如大佬所说，这里有很多权衡，pingpang操作中，假如compute加速了，与其并行运行的数据收发过程没有加速的话，整体也也不到多少提速作用，这样的话带宽就是影响速度的瓶颈。

hisrg, 2019-12-14:
请问楼上你的权值是怎么得到的？我这里只有一个权值文件，而hls源代码需要5个：
如下：
	FILE *fp_w = fopen("weightsv2_comb_reorg_ap16.bin", "rb");
    if(!fp_w) file_error("weightsv2_comb_reorg_ap16.bin");

	FILE *fp_b = fopen("biasv2_comb_ap16.bin", "rb");
    if(!fp_b) file_error("biasv2_comb_ap16.bin");
请告诉我这些权值下载位置，谢谢！！

dhm2013724, 2019-12-14:
> 请问楼上你的权值是怎么得到的？我这里只有一个权值文件，而hls源代码需要5个：
> 如下：
> FILE *fp_w = fopen("weightsv2_comb_reorg_ap16.bin", "rb");
> if(!fp_w) file_error("weightsv2_comb_reorg_ap16.bin");
> 
> ```
> FILE *fp_b = fopen("biasv2_comb_ap16.bin", "rb");
> if(!fp_b) file_error("biasv2_comb_ap16.bin");
> ```
> 
> 请告诉我这些权值下载位置，谢谢！！

跑一遍Software Version里的代码就全有了，兄弟！ Good Luck!

dhm2013724, 2019-12-14:
> > > 我觉得有个地方可以改进一下，算了一下weights大小大约97MB，FM来回读写的大小只有35MB左右，可是分配4给HP接口给FM用，只有一个HP给weights，如果要加速的话，可能这儿要重新分配一下
> > 
> > 
> > 是的，这个问题我想过；实际上，得先测下读写带宽的，最好是就4个HP，读IFM和W共享HP就可以了；带宽够也可以多加HP，或者直接接口换AXI-Stream+SG DMA；关于进一步改进，如果还是用HLS，这东西写不出很多结构，你可以先在软件版上改Tm Tn Tr Tc等等，然后等软件版那几部都对了，再移到HLS上去；软件版里幻数比较多，有很多trade off。。。现在的网络kernel都比较小了，一般就1_1和3_3，我不知道Winograd还是不是那么有效果，可能FFT会更好？或者试试Systolic Array也可以，这个HLS还是可以做的，之前看到Xilinx官方好像有example，这领域已经比较成熟了，玩玩可以，个人不建议跳坑，再往下学术贡献估计不大，Jason Cong都去搞存储了= =
> 
> 非常感谢大佬回复，通过学习这份代码学习到了不少有关hls的知识，感觉比看官网教程文档收获要大很多，谢谢大佬的无私奉献。我在软件这边修改了Tm Tn Tr Tc等等，这些影响weights的排布，然后怎么修改都可以运行出正确结果，说明软件这边没受到你说的幻数的影响，可是在fpga这边运行结果却不正确，我主要是修改Tm，把32改成128，这样乘法器就能用到512个，然后把和Tm有关的数据位宽加了两位，使用的是对应排布的weihts，这里想请教一下大佬，我这儿是不是有哪里疏漏的地方；

你看一下，可能是某些地方的位宽我设置的过小了，所以会产生这种问题，软件版没错的话，就看C仿真对不对了，C仿真也对的话，那么应该没事，不然得去找一下你改了哪些地方的位宽，那些地方我为了省资源一般都设的刚刚好，所以你改完就会这样；还有就是那个write back模块里面，考虑到yolov2最后一层，好像做了一些特定优化，所以我不知道你改完设计参数后，会不会也需要改；主要就看你最后C仿真的输出结果了，和软件版一致就OK了。 Good Luck！

hisrg, 2019-12-15:
请问能否提供这些文件？我使用vs 跑software version需要时间？我的邮件地址是afetsc2008@sina.com.谢谢

benpaodewali, 2019-12-16:
> 你看一下，可能是某些地方的位宽我设置的过小了，所以会产生这种问题，软件版没错的话，就看C仿真对不对了，C仿真也对的话，那么应该没事，不然得去找一下你改了哪些地方的位宽，那些地方我为了省资源一般都设的刚刚好，所以你改完就会这样；还有就是那个write [...]

C仿真会不会很慢，我今天试一下，位宽的话我都设置成uint试一下哈哈，谢谢大佬，出了结果再来汇报

benpaodewali, 2019-12-16:
> 请问能否提供这些文件？我使用vs 跑software version需要时间？我的邮件地址是[afetsc2008@sina.com](mailto:afetsc2008@sina.com).谢谢

你还需要不，100M左右

hisrg, 2019-12-16:
需要，谢谢

benpaodewali, 2019-12-17:
> > > > > 我觉得有个地方可以改进一下，算了一下weights大小大约97MB，FM来回读写的大小只有35MB左右，可是分配4给HP接口给FM用，只有一个HP给weights，如果要加速的话，可能这儿要重新分配一下
> > > > 
> > > > 
> > > > 是的，这个问题我想过；实际上，得先测下读写带宽的，最好是就4个HP，读IFM和W共享HP就可以了；带宽够也可以多加HP，或者直接接口换AXI-Stream+SG DMA；关于进一步改进，如果还是用HLS，这东西写不出很多结构，你可以先在软件版上改Tm Tn Tr Tc等等，然后等软件版那几部都对了，再移到HLS上去；软件版里幻数比较多，有很多trade off。。。现在的网络kernel都比较小了，一般就1_1和3_3，我不知道Winograd还是不是那么有效果，可能FFT会更好？或者试试Systolic Array也可以，这个HLS还是可以做的，之前看到Xilinx官方好像有example，这领域已经比较成熟了，玩玩可以，个人不建议跳坑，再往下学术贡献估计不大，Jason Cong都去搞存储了= =
> > > 
> > > 
> > > 非常感谢大佬回复，通过学习这份代码学习到了不少有关hls的知识，感觉比看官网教程文档收获要大很多，谢谢大佬的无私奉献。我在软件这边修改了Tm Tn Tr Tc等等，这些影响weights的排布，然后怎么修改都可以运行出正确结果，说明软件这边没受到你说的幻数的影响，可是在fpga这边运行结果却不正确，我主要是修改Tm，把32改成128，这样乘法器就能用到512个，然后把和Tm有关的数据位宽加了两位，使用的是对应排布的weihts，这里想请教一下大佬，我这儿是不是有哪里疏漏的地方；
> > 
> > 
> > 你看一下，可能是某些地方的位宽我设置的过小了，所以会产生这种问题，软件版没错的话，就看C仿真对不对了，C仿真也对的话，那么应该没事，不然得去找一下你改了哪些地方的位宽，那些地方我为了省资源一般都设的刚刚好，所以你改完就会这样；还有就是那个write back模块里面，考虑到yolov2最后一层，好像做了一些特定优化，所以我不知道你改完设计参数后，会不会也需要改；主要就看你最后C仿真的输出结果了，和软件版一致就OK了。 Good Luck！
> 
> C仿真有问题，直接退出，你说的write back我好像看到了和tm有关的幻数，最后一层有425个filter：
> `if(TM_MIN_6b==9) TM_MIN_g = 12; else TM_MIN_g = TM_MIN_6b; `
> 好像是因为425=32*13+9，后面我感觉TM_MIN_g=9应该也可以呀，大佬为啥在这儿变成12，是因为奇偶吗？然后想问一下大佬是怎么仿真调试的，我这边c simulation直接死掉了。如果上板每次改一下重新综合加上编译要大半个小时。。
> 用tb里面的teshbench测试，把ap_int<>这类数据都用int和unsigned int代替，结果128也是正确的

尝试Tm=128，使用tb里面的代码测试，把ap_uint类型都换成uint，结果正确，这样应该可以确定hls这边逻辑没有问题，继续综合到fpga里还是不对，那么应该推断问题出在sdk这边，我对比一下sdk里面的代码和tb里的代码的区别

benpaodewali, 2019-12-18:
> 尝试Tm=128，使用tb里面的代码测试，把ap_uint类型都换成uint，结果正确，这样应该可以确定hls这边逻辑没有问题，继续综合到fpga里还是不对，那么应该推断问题出在sdk这边，我对比一下sdk里面的代码和tb里的代码的区别

还是这个问题，这次使用hls的c simulation仿真，Tm=128的情况下，把所有ap_uint和ap_int位宽拓宽两位，结果正确，考虑到Cosimulation仿真需要很多时间，就直接上板子，结果还是不对，检查了一下，第一层的输出就是乱的，头疼不知道为啥

benpaodewali, 2019-12-19:
> 你看一下，可能是某些地方的位宽我设置的过小了，所以会产生这种问题，软件版没错的话，就看C仿真对不对了，C仿真也对的话，那么应该没事，不然得去找一下你改了哪些地方的位宽，那些地方我为了省资源一般都设的刚刚好，所以你改完就会这样；还有就是那个write [...]

大佬有微信吗，或者上次的微信群能不能分享一下，想请教一下哦，谢谢了

wpyueyue, 2019-12-20:
楼上的大佬，vivado创建的工程，最后地址分配是什么样子的。我自动布线，然后验证了，但是地址报错，我只能自己分配地址，但是我感觉有问题，想问以下您是怎样分配的。因为想在linux上移植，但是vivado地址有问题，所以没有向下进行。

[image]

wpyueyue, 2019-12-20:
[image]

benpaodewali, 2019-12-24:
> 楼上的大佬，vivado创建的工程，最后地址分配是什么样子的。我自动布线，然后验证了，但是地址报错，我只能自己分配地址，但是我感觉有问题，想问以下您是怎样分配的。因为想在linux上移植，但是vivado地址有问题，所以没有向下进行。
> 
> [image]

hp接口全部设置成0到1G，gp接口64k，你的没错

benpaodewali, 2019-12-24:
> 你看一下，可能是某些地方的位宽我设置的过小了，所以会产生这种问题，软件版没错的话，就看C仿真对不对了，C仿真也对的话，那么应该没事，不然得去找一下你改了哪些地方的位宽，那些地方我为了省资源一般都设的刚刚好，所以你改完就会这样；还有就是那个write [...]

大佬，我把weight分配到4个hp接口了，120M频率下，TM=128，乘法器使用了512个，一张图片0.633s，没有预想的快哦

closed 2020-01-03 by benpaodewali