dhm2013724/yolov2_xilinx_fpga #56: 关于biasv2_comb_ap16.bin文件的疑问？

Microdent, 2019-12-19:
你好，近期我在把这个项目移植到ZCU102上面，但是最后在板子上跑的时候，每次执行到copy_file2mem("biasv2_comb_ap16.bin",(43044+4)/2,BETA_BASE);（yolov2.h 的yolov2_hls_ps()函数中）这一句的时候总是会出现bus error的错误，但是执行前面的copy_file2mem("weightsv2_comb_reorg_ap16.bin",(203767168)/2,WEIGHT_BASE);并不会报错，然后逐步分析，发现是最后是copy_mem2dev()函数中的memcpy(virtual_addr,orig,byte_num);这句话会产生错误。
我一直想不出为什么会出现这样的错误，毕竟之前拷贝weights的时候也可以正常进行，而且我也在设备树分配了256M内存（地址是0x6000_0000-0x7000_0000），然后设置了WEIGHT_BASE = 0x60000000，BETA_BASE = 0x6C25F000，MEM_BASE  = 0x1C26A000，请问你之前是否遇到类似问题，能否提供一下解决的思路呢？

dhm2013724, 2019-12-19:
> 你好，近期我在把这个项目移植到ZCU102上面，但是最后在板子上跑的时候，每次执行到copy_file2mem("biasv2_comb_ap16.bin",(43044+4)/2,BETA_BASE);（yolov2.h 的yolov2_hls_ps()函数中）这一句的时候总是会出现bus error的错误，但是执行前面的copy_file2mem("weightsv2_comb_reorg_ap16.bin",(203767168)/2,WEIGHT_BASE);并不会报错，然后逐步分析，发现是最后是copy_mem2dev()函数中的memcpy(virtual_addr,orig,byte_num);这句话会产生错误。
> 我一直想不出为什么会出现这样的错误，毕竟之前拷贝weights的时候也可以正常进行，而且我也在设备树分配了256M内存（地址是0x6000_0000-0x7000_0000），然后设置了WEIGHT_BASE = 0x60000000，BETA_BASE = 0x6C25F000，MEM_BASE = 0x6C26A000，请问你之前是否遇到类似问题，能否提供一下解决的思路呢？

为了防止段错误，这边我记得是类似按4KB大小还是多少KB大小的整数页拷贝的；按道理讲，这里是不会错的，你在函数里加一句printf，输出一下每次的起始地址和byte_num看一下

Microdent, 2019-12-19:
是按照4k整数倍拷贝的，我输出过起始地址和byte_num，都没有任何问题，而且设备地址virtual_addr也能正常申请下来，就是执行memcpy报bus error，而且我还尝试过改变weights和bias的拷贝顺序，改变weights和bias内存地址的先后顺序，bias都不能正常拷贝，weights都可以正常拷贝，而且bias只是一个22k的小文件，我现在也挺绝望的🌚🌚🌚

dhm2013724, 2019-12-19:
> 是按照4k整数倍拷贝的，我输出过起始地址和byte_num，都没有任何问题，而且设备地址virtual_addr也能正常申请下来，就是执行memcpy报bus error，而且我还尝试过改变weights和bias的拷贝顺序，改变weights和bias内存地址的先后顺序，bias都不能正常拷贝，weights都可以正常拷贝，而且bias只是一个22k的小文件，我现在也挺绝望的🌚🌚🌚

是最后一次传bias时报错，还是一开始第一次传就报错？

Microdent, 2019-12-19:
每次是传最大1024*1024byte大小的块，bias只有22k，所以只需要传一次，出错的就是这一次，weights有98M左右所以要传98次，weights的98次都是正常的

dhm2013724, 2019-12-19:
> 每次是传最大1024*1024byte大小的块，bias只有22k，所以只需要传一次，出错的就是这一次，weights有98M左右所以要传98次，weights的98次都是正常的

你现在是最后一次传22K还是1MB？试一下传32KB

Microdent, 2019-12-19:
最后一次是传22k，因为bias不满1M，我明天而且由于22k实际上不是4k的整数倍，所以最终穿的时候是比bias真实大小要大一些的（补齐4k），所以我觉得尝试32k可能意义不大，我觉得是不是我生成的bias文件有问题？但是跑hls的c simulation又是正常的

Microdent, 2019-12-19:
最后一次是传22k，因为bias不满1M，而且由于22k实际上不是4k的整数倍，所以最终穿的时候是比bias真实大小要大一些的（补齐4k），所以我觉得尝试32k可能意义不大，我觉得是不是我生成的bias文件有问题？但是跑hls的c simulation又是正常的

Microdent, 2019-12-19:
最后一次是传22k，因为bias不满1M，而且由于22k实际上不是4k的整数倍，所以最终传的时候是比bias真实大小要大一些的（补齐4k），所以我觉得尝试32k可能意义不大，我觉得是不是我生成的bias文件有问题？但是跑hls的c simulation又是正常的

dhm2013724, 2019-12-19:
> 最后一次是传22k，因为bias不满1M，而且由于22k实际上不是4k的整数倍，所以最终传的时候是比bias真实大小要大一些的（补齐4k），所以我觉得尝试32k可能意义不大，我觉得是不是我生成的bias文件有问题？但是跑hls的c simulation又是正常的

我又看了一下代码，你看一下会不会是位宽的原因

Microdent, 2019-12-19:
具体是指什么位宽呢？ZYNQ ip核上的AXI总线位宽吗？ZCU102的ZYNQ UltraScale+ MPSoC的axi总线最高支持128位位宽，但是为了和你用的ZYNQ 7000系列的ip核上的64位位宽相匹配，我就也设置成了64，不知道是否可能是这个原因？

dhm2013724, 2019-12-19:
> 具体是指什么位宽呢？ZYNQ ip核上的AXI总线位宽吗？ZCU102的ZYNQ UltraScale+ MPSoC的axi总线最高支持128位位宽，但是为了和你用的ZYNQ 7000系列的ip核上的64位位宽相匹配，我就也设置成了64，不知道是否可能是这个原因？

用原来的默认位宽，问题应该在这

Microdent, 2019-12-19:
嗯嗯，我明天试试默认位宽，感谢大佬这么晚还在帮我分析问题，非常感谢！

Microdent, 2019-12-20:
> > 具体是指什么位宽呢？ZYNQ ip核上的AXI总线位宽吗？ZCU102的ZYNQ UltraScale+ MPSoC的axi总线最高支持128位位宽，但是为了和你用的ZYNQ 7000系列的ip核上的64位位宽相匹配，我就也设置成了64，不知道是否可能是这个原因？
> 
> 用原来的默认位宽，问题应该在这
我刚刚把axi总线改成了默认位宽
[image]
但是还是产生bus error的错误
[image]
[image]
可以看出weights是被正确拷贝的，但是一到22k的bias就会在memcpy那里报错，由输出结果来看输出的rd_num=21524和RequestByteNum=24576都是正确的，in_buffer=1814425600=0x6c25f000与我分配的内存地址也是一致的
[image]
通过输出内存信息也可以看到0x6000_0000-0x7000_0000段的256M内存确实为我保留了
[image]
实在是想不出为什么会在这个地方报错了

hisrg, 2019-12-20:
你们没在pynq上跑吗？

Microdent, 2019-12-26:
另外我想问一下SDK/xconv_hw.h中的#define YOLO2_BASEADDR 0x43c00000，这个地址是自定义的吗，还是有什么依据呢？我在运行到YOLO2_FPGA函数中xbase_address = (unsigned char *)mmap(NULL, map_len, PROT_READ | PROT_WRITE, MAP_SHARED, fd, (off_t)PhysicalAddress);这句时出现了operation not permitted的错误

dhm2013724, 2019-12-27:
> 我刚刚把axi总线改成了默认位宽
> [image]
> 但是还是产生bus error的错误
> [image]
> [image] [...]

我今天在ZCU102上测试了一下，开始也出现了你说的问题，然后我仔细看了一下，是copy weight from file to mem的时候，最后一段报错，我改了一下，让最后一段多copy一些字节，也copy PAGESIZE大小，然后就过去了
[image]
[image]
[image]
ZCU102版的代码，这段时间我也更新到git上来，基本就写数据那边改了一点

Microdent, 2019-12-27:
> 我今天在ZCU102上测试了一下，开始也出现了你说的问题，然后我仔细看了一下，是copy weight from file to mem的时候，最后一段报错，我改了一下，让最后一段多copy一些字节，也copy PAGESIZE大小，然后就过去了
> [image]
> [image]
> [image]
> ZCU102版的代码，这段时间我也更新到git上来，基本就写数据那边改了一点

嗯嗯是的，我前几天手动在bias后面加了些无效数据就可以跑了，一直忘了更新这个issue，感谢大佬！

closed 2019-12-27 by dhm2013724