dhm2013724/yolov2_xilinx_fpga #63: "Effective bandwidth for 64MB transmission"

ms0488638, 2020-02-20:
Hi,

I have read your article, and noticed there is a line graph about the effective bandwidth.

[image]

I wonder if this is tested on PYNQ, and how is the setup(block design and software

It would appreciate if there is further detail about this test.

Thank you.

dhm2013724, 2020-02-20:
> Hi,
> 
> I have read your article, and noticed there is a line graph about the effective bandwidth.
> 
> [image]
> 
> I wonder if this is tested on PYNQ, and how is the setup(block design and software
> 
> It would appreciate if there is further detail about this test.
> 
> Thank you.

You can write a funciton in HLS that has three arguments: one pointer for AXI4-Master interface, one integer type variable data_num to write/read data_num through AXI4-Lite Slave interface, and one integer type variable burst_length to limit the max burst transmission length. Just use HLS to design a simple DMA ip, and set different busrt length(1~256) and data_num to evaluate the effect bandwidth.

ms0488638, 2020-02-24:
Thanks, this solve my question.

One more thing, in ug585, the maximum bandwidth shows the AXI_HP is 1200 MB/s.

However, your result says its only 83% of maximum bandwidth, which means the maximum bandwidth is about 600 MB/s.

How come?