dhm2013724/yolov2_xilinx_fpga #73: 您好，我遇到了一些问题，想加一下您的微信讨论群可以吗?

xbdxwyh, 2020-07-12:


kangyuzhe666, 2020-07-13:
上通

kangyuzhe666, 2020-07-13:
上同

clancylea, 2020-07-13:
可以留下微信号，拉你进群

kangyuzhe666, 2020-07-13:
非常感谢您，我的微信号：13777453014

发自我的iPhone

Rubick-fafafa, 2020-07-15:
@clancylea 你好 可以拉我进微信群吗 微信号：wx19801219863

xuemo188, 2020-08-02:
> 可以留下微信号，拉你进群

大佬我的微信：18811019716，求进群

mq1h, 2020-09-10:
微信号：Mingqiang_Huang，希望能进群

noobgrow, 2020-09-17:
@xuemo188 您好！微信号17356510926，求进群

AliceInCh, 2020-09-24:
微信号L1187734347，求进群

jackZ48, 2020-10-16:
微信号：zsb421023  求进群！

tzuj6, 2020-11-25:
您好 我們有兩個人 想進群
wechat ID: tzuj_6, shane_5713
謝謝!

kangyuzhe666, 2020-11-25:
我的微信 13777453014

发自我的iPhone

LinErDeng, 2020-12-10:
您好，同款想进微信群，微信号：LinEr_Deng

kangyuzhe666, 2020-12-10:
你好加我wx拉你进群
wx：13777453014

发自我的iPhone

LeiWang1999, 2021-01-03:
hello，wx_id: wanglei199901
Thanks!

zhaobr340104, 2021-04-08:
您好,我们有两个人想进群！
微信号zbr330232872、W1491040770，求进群，谢谢！

kangyuzhe666, 2021-04-08:
请加微信 137774583014 拉你入群

发自我的iPhone

xiying-boy, 2021-06-23:
您好,我也想进群,请问还能拉我一下吗?微信号18428321561

FoolgryGamer, 2021-07-14:
您好，我也想进群，微信号w18888920974，十分感谢

Baymaxly, 2021-07-23:
您好，我也想入群交流一下，请问还能拉一下吗，十分感谢！微信18370752330

evanlee2021, 2021-12-01:
请问这个微信群还在吗？ 我也想进群 ，微信号：wznnt1987

zhaijiaqi, 2022-01-26:
这个群还在么？求大佬拉我进群：zjq1215811790

FoolgryGamer, 2022-01-26:
***@***.***

Gengchen2018, 2022-03-13:
请求进群，18662637257

FoolgryGamer, 2022-03-13:
***@***.***

kangyuzhe666, 2022-03-13:
加我微信 13777453014

FoolgryGamer, 2022-03-28:
***@***.***

Alooonso, 2022-04-20:
long2578709784 求拉

FoolgryGamer, 2022-04-20:
***@***.***

yuansenhao, 2022-04-25:
我的微信号 glowing_sun 求拉，有个问题困扰好久了

JinlinYeJinlinYe, 2022-09-27:
Jinlin_leaf我的微信号，求拉

FoolgryGamer, 2022-09-27:
***@***.***

FoolgryGamer, 2022-10-25:
***@***.***

TSeminent, 2023-03-15:
> 可以留下微信号，拉你进群

大佬您好，这是我的微信号：redtay1989   求拉进群

FoolgryGamer, 2023-03-15:
***@***.***

TSeminent, 2023-03-15:
> 

您好，现在还能进群吗，我的微信号：redtay1989

TSeminent, 2023-03-15:
> 请加微信 137774583014 拉你入群 发自我的iPhone
> […](#)

您好，现在还能进群吗，我的微信号：redtay1989

YINTIANYY, 2023-05-25:
您好现在还可以进群吗？我的微信号：15370861879

218w1d7706, 2023-07-17:
@YINTIANYY hey did u joined the group

FoolgryGamer, 2023-07-17:
***@***.***

YINTIANYY, 2023-08-29:
我的VX是15370861879，可以拉我进去吗？我想问一下有关量化的问题

FoolgryGamer, 2023-08-29:
***@***.***

jieun0904, 2023-11-27:
我想加入这个小组。请  ID: wxid_okp4iwkjvsa912

FoolgryGamer, 2023-11-27:
***@***.***

YinChinhao, 2024-02-25:
我的wx是TdkrLee,能拉我进群吗？