dhm2013724/yolov2_xilinx_fpga #76: C仿真的latency不确定

lovvtin, 2020-12-16:
你好，我对src_int16下的cnn.cpp进行C仿真，latency不确定，请问这是什么原因呢

dhm2013724, 2020-12-16:
You should add macro to tell the compiler that every uncertained loop's tripcount. Otherwise, HLS could only assume some simple and short variables' value.

lovvtin, 2020-12-18:
Thanks, the problem about latency has been solved. But I wonder why there is error, "Aborting co-simulation: top function 'YOLO2_FPGA' is not invoked in the test bench", when  I run C/RTL cosimulation.

qing-2, 2022-03-28:
> 请问是怎么解决的呢？ 要在哪些地方给哪些变量添加宏定义呀？

This may help you. For example `DO_PRAGMA(HLS LOOP_TRIPCOUNT min=1 max=Tc)` at loop body.
[https://docs.xilinx.com/r/en-US/ug1399-vitis-hls/pragma-HLS-inline](https://docs.xilinx.com/r/en-US/ug1399-vitis-hls/pragma-HLS-inline)

```cpp
#define PRAGMA_SUB(x) _Pragma (#x)
#define DO_PRAGMA(x) PRAGMA_SUB(x)
```