dhm2013724/yolov2_xilinx_fpga #81: Some files do not have a repo.  ( ...  .bin)

muratozcn, 2021-01-17:
Hello, congratulations on your work. While trying to create the IP Block I noticed some files were missing. I marked the missing files in the picture below. There is some information for creating these files. But I could not. If possible, can you send these files to me?

[https://yadi.sk/i/POnInHnRn5wm-Q](url)

dhm2013724, 2021-01-17:
Current project has been updated several times, but the img didn't changed. Some bin files about weight and quantization are generated from the repo software version. So, you should first exec some steps in this repo to generate the bin files that you need.

muratozcn, 2021-01-17:
yolov2 xilinx fpga / hls /
I think I will create it using the files here. But I couldn't find enough information on how to create weights and quantities. Can you give more detailed information on how to create weights and quantities?

dhm2013724, 2021-01-18:
> yolov2 xilinx fpga / hls /
> I think I will create it using the files here. But I couldn't find enough information on how to create weights and quantities. Can you give more detailed information on how to create weights and quantities?

Yep, you should first read and follow the README in top repo/software version, follow the step 1 and step 2. In step 1, you can get the original weight and bias bin file combined with batch normalization. In step 2, you will get the bin files about reorder and quantization. OK, if you have completed previous steps, you will get the files that you need in hls.

muratozcn, 2021-01-18:
I understood what you said. But I am having problems with the second and third steps. I could not perform the second step. Is there a place where I can learn more detailed information on how to perform the second and third steps?

YINTIANYY, 2023-08-29:
> 我明白你说的话。但是我在第二步和第三步方面遇到了问题。我无法执行第二步。有没有一个地方可以让我了解有关如何执行第二步和第三步的更多详细信息？
你好，请问你解决这个问题了吗？我也卡在第二部了，不明白我应该如何执行