dhm2013724/yolov2_xilinx_fpga #87: 关于权重文件.bin生成的问题

starkLJ, 2021-05-18:
大佬您好，我在150M的分支里的software_version中按照步骤生成了bias_ap16.bin等文件，但是在yolov2.h中用到的是biasv2_comb_ap16.bin等文件，我尝试改名替换，但是运行结果是有错的，请教一下怎么生成上述文件

gexianwei1231, 2021-05-21:
您好，请问一下Windows中怎么生成bin文件，麻烦您解答一下，感谢

starkLJ, 2021-05-21:
按照分支下software_version做,如果是windows，查一下windows搭建yolo环境，最好还是在Linux环境下搭建

gexianwei1231, 2021-05-22:
这个bin文件有什么作用，我没太明白

starkLJ, 2021-05-22:
大概就是将原有的权重进行量化，量化成fpga可以用的权重文件，他的工程下分成了int和float，两种不同的量化。本质上还是和他fpga 底层的hls代码有关系，他这么做是为了方便他用量化后的文件，如果你自己写hls，可以按你自己的想法量化文件。详细的内容你可以参考他的硕士论文

daviejones, 2021-06-24:
Hello @starkLJ, I am having the same the issue regarding the incorrect results after generating the weight and bias files using the methods from the software_version. Were you able to resolve this issue? Also, are you working with the 16-bit or 32-bit fp version?

Thanks in advance

a294981774, 2021-06-30:
> 大佬您好，我在150M的分支里的software_version中按照步骤生成了bias_ap16.bin等文件，但是在yolov2.h中用到的是biasv2_comb_ap16.bin等文件，我尝试改名替换，但是运行结果是有错的，请教一下怎么生成上述文件

您好，请问您生成的bin文件能在pynq上面运行吗？我运行出来的都是框框

starkLJ, 2021-07-02:
没有试过pynq，如果权重没有问题的话，你运行出来都是框框的话多半是内存的问题，有可能是地址错误之类，可以按照这个思路排查一下