dhm2013724/yolov2_xilinx_fpga #92: 您好！請問 IP 輸出 ports 問題 ！

duchungk7, 2021-06-24:
我跟著流程對 建立 HLS IP 用於 PYNQ, 
但 IP ports 輸出 跟您描述的 IP 電路不一樣 （如圖）
請問是我流程有問題嗎？還是什麼原因呢？
[image]

TSeminent, 2023-03-05:
您好，请问您问题解决了吗？我也遇到了同样的问题

TSeminent, 2023-03-05:
> 我跟著流程對 建立 HLS IP 用於 PYNQ, 但 IP ports 輸出 跟您描述的 IP 電路不一樣 （如圖） 看您的 IP 包含 5 個 DATA_BUS 輸出，我這邊只有兩個 請問是我流程有問題嗎？還是什麼原因呢？ [image]

您好，请问您问题解决了吗？我也遇到了同样的问题

gb16001, 2023-12-30:
我的测试是，这个repo的flex分支没有一个hls代码综合出来的IP是5个M AXI总线的

影响M AXI接口的是这个定义接口行为的宏： 因为只出现过两个代号所以hls综合出来的IP只会有两个M AXI口
[image]
[image]

khoinguyenph, 2024-04-20:
So can you show us how to solve?
Thanks you

khoinguyenph, 2024-04-20:
> 我跟著流程對 建立 HLS IP 用於 PYNQ, 但 IP ports 輸出 跟您描述的 IP 電路不一樣 （如圖） 看您的 IP 包含 5 個 DATA_BUS 輸出，我這邊只有兩個 請問是我流程有問題嗎？還是什麼原因呢？ [image]

Help me with the same problem please sir.

gb16001, 2024-04-21:
> 我的测试是，这个repo的flex分支没有一个hls代码综合出来的IP是5个M AXI总线的
> 
> 影响M AXI接口的是这个定义接口行为的宏： 因为只出现过两个代号所以hls综合出来的IP只会有两个M AXI口 

M AXI ports is defined in hls macro. there is only 2 maxi bundle id in the code, so only 2  ports will be synthesized out.to fix this, you can bound parameters to different id or just ignore this. this may reduce ram io performance but IP should still works.
![image](https://private-user-images.githubusercontent.com/87075578/293452260-2176a03a-e3cc-44a9-bf63-d01ada3e1c49.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM2NzUzMDYsIm5iZiI6MTcxMzY3NTAwNiwicGF0aCI6Ii84NzA3NTU3OC8yOTM0NTIyNjAtMjE3NmEwM2EtZTNjYy00NGE5LWJmNjMtZDAxYWRhM2UxYzQ5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDIxVDA0NTAwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJlNjczZTE4NDAzYTE0YmMyMzQ1ZDIzYWY0MjI4YmM2ZmUwMGQwYTQ1MDFlNzcyYTYzY2Q3NTkzNWQyYzAwZGYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.3yPVdNFbjtFx1i2uL0ZcvxI1Xnn2FkTxCmfPbwBYQ5s) ![image](https://private-user-images.githubusercontent.com/87075578/293452401-e6467e5d-427b-408e-b2d1-dfe1ced17548.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM2NzUzMDYsIm5iZiI6MTcxMzY3NTAwNiwicGF0aCI6Ii84NzA3NTU3OC8yOTM0NTI0MDEtZTY0NjdlNWQtNDI3Yi00MDhlLWIyZDEtZGZlMWNlZDE3NTQ4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDIxVDA0NTAwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQwZGY3MDllMDk0MjI2YmNlNGYwMDk2YTg5YzAzZDNjZGQzNWQwNTJmZTllN2E1MTgzNzQ5NDYzZTgwNmJiOWQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.QHMC-oNT1BvvO-LG6RJBiVLg5rakhoYbbK3E9xwv5bU)