dhm2013724/yolov2_xilinx_fpga #98: No Errors but Blank Output Feature Maps

swleungbrian, 2021-11-10:
Dear Sir,

I am trying the pynq notebook and everything went fine without errors.
Yet, it cannot detect anything and output feature maps are always blank.

Actually, i am using ax7020, which is quite simiar to Pynq-z2. 
Not sure if you could advise me on this.

Regards,

Brian

qing-2, 2022-03-28:
The detection is written by the python part.

At first I used my own .bit file,the output is the original picture without detection boxes.
Finally I used the author's .bit file and .hwh file I generated by myself（used the author's hls code）,everything goes well.

gb16001, 2024-01-15:
i faced  this problem too. it is now fixed by simply set s axi hp0 data width to 64 instead of 32
[image]

for more deails please chack this https://github.com/dhm2013724/yolov2_xilinx_fpga/issues/125#issuecomment-1891522666 ( in chinese)