dhm2013724/yolov2_xilinx_fpga #113: 关于硬件的reorg模块

ExcitingYi, 2023-02-20:
作者您好，我在pynqz2上成功部署过您之前Tn4Tm32的版本，并对模型其进行剪枝，改进了hls的计算模块，取得了加速效果。

但我想进一步改进您的方案时遇到了一些问题：前几个版本的reorg是在硬件中实现的（在hls中有一个reorg_yolo2电路模块）。但flex分支更新之后，把reorg模块给放到了cpu上实现。SDK代码更新了，但pynq的顶层代码还没有更新。

我尝试去模仿sdk的c代码写pynq顶层的python代码，但是python的接口不知道怎么从fpga的片上内存调取数据（即需要reorg的feature maps的数据）。也去尝试把以前的reorg_yolo2的hls移植过来，但不理解reorg过程数据读取和存储（好像和Tn4Tm32耦合很紧密，无法移植），所以也没有成功。

目前还未解决这个问题，希望能获得作者的联系方式进一步沟通，谢谢！

dhm2013724, 2023-02-20:
PYNQ部分原来是@clancylea根据SDK C代码写的，他现在应该不维护了；上周末我更新了下float32类似Tn4Tm36 4输入2输出并发的c/c++版本，如果你能在上面改的话，可以先试一下，等周末我再来更新PYNQ的部分，周一到五我应该只会更新int16部分;reorg主要latency基本可以忽略，所以就直接放CPU做了，主要PYNQ上跑python还是太慢了，得把所有的c/c++都用cpython包起来直接调用比较好

ExcitingYi, 2023-02-20:
感谢回复！我继续接着尝试reorg的hls代码。期待更新。

ExcitingYi, 2023-03-07:
这边这个问题已经解决了。

解决方法是改hls的电路模块，把之前版本的reorg模块给拿来用。有些踩坑的地方给大家提醒一下：
1. 需要reorg处理的时候顶层pynq代码的output_w，output_h，tm，tn，都是很巧妙得处理过的，具体怎么巧妙还得结合tn4tm32版本的hls还有pynq代码自己理解。
2. 需要reorg的output feature map是13x13，但做reorg的是26x26为单位的。也就是四张feature map一起做。（13不能被4整除）
3. tm32tn4版本为了追求效率，原先版本的13x13的feature map都被很巧妙得处理成26x26的传输了（牺牲了灵活性，所以是tm32tn4定死了）然后现在这个flex版本，13x13的feature map需要被处理成13x14，这就导致需要reorg时候，存output feature map的操作需要做一定修改。
4. pynq也需要更改，主要体现在各种配置上，因为现在还没更新的pynq代码的网络配置（比如net_layers_output这个配置）都是按照原先13x13的feature map的偏置，但现在都应该改成13x14的配置。

debug了两周，从软件de到硬件，终于搞出来了。。。暂时总结了这四个坑。

现在回过头来看，改hls代码真的不如直接按着c语言的sdk改pynq顶层方便。因为但当时不理解in_ptr，out_ptr怎么从img_base_buffer里面调数据。就想着改hls。。。结果到头来，因为debug需要，还是把inptr outptr img_base_buffer给理解了一遍，麻了。。但至少收获也很大。。。。

closed 2023-03-07 by ExcitingYi