dhm2013724/yolov2_xilinx_fpga #114: 关于HLS输出的IP的端口问题，端口数与图片不匹配

TSeminent, 2023-03-05:
我的IP核右侧只有两个BUS，图片里的IP有5条BUS，这是什么原因呢

TSeminent, 2023-03-05:
[image]

ExcitingYi, 2023-03-07:
因为作者用的图片是以前n4m32那个版本的。现在综合的是flex版本的。但本质都是一样的。按照给的block design找猫画虎就行。我这里有一份你可以参考一下。
[image]

TSeminent, 2023-03-13:
> 因为作者用的图片是以前n4m32那个版本的。现在综合的是flex版本的。但本质都是一样的。按照给的block design找猫画虎就行。我这里有一份你可以参考一下。 [image]

非常感谢

TSeminent, 2023-03-13:
> 因为作者用的图片是以前n4m32那个版本的。现在综合的是flex版本的。但本质都是一样的。按照给的block design找猫画虎就行。我这里有一份你可以参考一下。 [image]

您好，可以再问一下您这个图是对应float32的还是int16的呢？

ExcitingYi, 2023-03-13:
这个是int16的ip核，float32应该也是一样的。

TSeminent, 2023-03-13:
> 这个是int16的ip核，float32应该也是一样的。
好的，非常感谢