dhm2013724/yolov2_xilinx_fpga #117: 关于量化为int16后生成的bin文件问题

TSeminent, 2023-03-13:
作者在hls repo的README_reserved.md中说testbench需要用到5个bin文件，为什么我在software_version阶段生成的bin文件有6个，并且和作者截图里的bin文件名称不一样呢？以下是我生成的bin文件，想求教一下大家是生成错了，还是只需用6个中的5个文件？
[image]

TSeminent, 2023-03-13:
这里是作者截图里的bin文件：
[image]

changmushu, 2023-08-14:
> > 项目迭代很久了但是readme没有更新，需要在旧的fork分支里寻找到一个特殊版本即可对应这个readme
> 
> 你好。请问需要哪个版本呢？

你应该可以从我的star中找到，我star了两个分支，但是我不确定是哪一个有效（😂

sticktotheend, 2023-08-14:
> > > 项目迭代很久了但是readme没有更新，需要在旧的fork分支里寻找到一个特殊版本即可对应这个readme
> > 
> > 
> > 你好。请问需要哪个版本呢？
> 
> 你应该可以从我的star中找到，我star了两个分支，但是我不确定是哪一个有效（😂

谢谢大佬！！！