dhm2013724/yolov2_xilinx_fpga #119: makefile生成的Yolov2复制到SD卡，在板子上./Yolov2运行之后，运行到Layer[ 0]就停止了，请问是什么原因呢？

M-zizixixi, 2023-03-31:
Layer[ 0]: outputMemory: 5537792;BN=1;Activation=7;conv     32  3 x 3 / 1   416                                                                                                              x 416 x   3   ->   416 x 416 x  32  0.299 BFLOPs
板端跑到这一步就停止了

M-zizixixi, 2023-03-31:
[image]

M-zizixixi, 2023-03-31:
[image]

M-zizixixi, 2023-03-31:
已解决，是depth的问题，需要修改depth如下：
[image]

TSeminent, 2023-04-11:
> 已解决，是depth的问题，需要修改depth如下： [image]

您好，我也遇到了同样的问题，可以问一下您是在哪个文件里面更改吗，更改后的正确值是您红框内的数值吗

TSeminent, 2023-04-13:
> 已解决，是depth的问题，需要修改depth如下： [image]

您好，还能问一下这个数字是怎么算的吗，我用的精度是int16，max_read_burst_length、max_write_burst_length的数值和您图里的256不一样，所以不敢直接照搬您红框里的数值，想问一下您得到正确数值的方法

TSeminent, 2023-04-14:
> > 已解决，是depth的问题，需要修改depth如下： [image]
> 
> 您好，还能问一下这个数字是怎么算的吗，我用的精度是int16，max_read_burst_length、max_write_burst_length的数值和您图里的256不一样，所以不敢直接照搬您红框里的数值，想问一下您得到正确数值的方法

已解决，是reserved memory分配的问题

MWHYNOT, 2023-07-09:
@TSeminent ,大佬求问reserved memory分配在哪里改

Rayiy, 2023-12-24:
@MWHYNOT 您好，请问您解决这个问题了吗

218w1d7706, 2024-07-22:
@TSeminent help me bro. how to calculate this depth value