dhm2013724/yolov2_xilinx_fpga #125: pynq文件夹中的bit file是那个版本的hls的release？which edition fo hls code can build the bit file released  in pynq folder?

gb16001, 2023-12-30:
我正在复现这个repo，最后py脚本卡在了IP驱动函数YOLO__Init_EX
## 环境
board:pynq-z2
IDE: vivado 2023.2 for overlay
Vitis HLS 2023.2 for synthesis ip core
hls代码使用的是flex分支的src_int16文件夹中的cnn_t.h、cnn_hls.cpp两个文件。
## 复现步骤
完成综合IP，实现overlay得到bitstream后上板使用ipynb脚本进行测试（脚本和权重文件已置于同一文件夹）。
## 运行结果
卡在第一个conv中。手动打断程序发现陷入了IP驱动函数的一个查看IP是否完成运算的循环
```
Open pictrue success!
pictrue size: (640, 425)
pictrue mode: RGB
0 conv
---------------------------------------------------------------------------
KeyboardInterrupt                         Traceback (most recent call last)
Input In [29], in <cell line: 16>()
      7 #frame_in_img = PIL_Image.fromarray(frame_in) 
      8 # img_boxed = letterbox_image(frame_in,416,416)
      9 
     10 # img_array_3_416_416 = image_to_array_1dim(img_boxed,416,416)
     11     
     12 # input_tmp_img = float32_int(img_array_3_416_416)
     14 img_out = frame_in
---> 16 yolo_meminout(frame_in,img_w,img_h,img_out)
     17 img_out

Input In [16], in yolo_meminout(frame_in, img_w, img_h, frame_out)
     30 start_time = time.time()
     32 region_buff = np.zeros((73008,), dtype=np.float32)
---> 33 yolo_fpga(img_base_buffer,region_buff)
     35 end_time = time.time()
     36 fpga_process_time = end_time - start_time 

Input In [9], in yolo_fpga(img_base_buffer, region_buff)
     48 if(i==26):
     49     INPUTQ_t = inputQ[12];
---> 51 YOLO__Init_EX(int(in_ptr[i]),int(out_ptr[i]),woffset//2,boffset//2,
     52     layerc[i],layern[i],layersize[i],
     53     layerstride[i],layerw[i],layerh[i],output_w,output_h,
     54     layerpad[i],layeractivation[i],layerbatch_normalize[i],
     55     TM,TN,TR,TC,
     56     mLoops,nLoops,rLoops,cLoops,0,
     57     INPUTQ_t,inputQ[offset_index+1],weightQ[offset_index],betaQ[offset_index],
     58     WEIGHT_BASE,BETA_BASE)
     60 woffset += weight_offset[offset_index]
     61 boffset += beta_offset[offset_index]  

Input In [23], in YOLO__Init_EX(In_Address, Out_Address, Weight_offset, Beta_offset, InFM_num, OutFM_num, Kernel_size, Kernel_stride, Input_w, Input_h, Output_w, Output_h, Padding, IsNL, IsBN, TM, TN, TR, TC, mLoops, nLoops, rLoops, cLoops, LayerType, InputQ, OutputQ, WeightQ, BetaQ, WEIGHT_BASE, BETA_BASE)
    113 mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_AP_CTRL,1)
    114 while True:
--> 115     ap_done =  (mmio.read(XYOLO2_FPGA_CTRL_BUS_ADDR_AP_CTRL)>>1)&0x01
    116     if(ap_done):
    117         break

KeyboardInterrupt:
```
## debug思路
推测是py脚本中的IP驱动函数YOLO__Init_EX和我自行综合的IP core不匹配。
查看了一下s axi的地址映射，发现和YOLO__Init_EX中的地址有一定的出入。故想询问一下**在pynq文件夹中发布的bin,py,bit文件对应的hls代码是那个版本？**
[image]

pgq18, 2024-01-08:
你好，请问这个问题解决了吗？我最近也在复现这个项目，也遇到同样的问题，可以一起交流一下吗？

gb16001, 2024-01-11:
目前我没能解决这个问题，这个仓库的复现在我这搁置了。
我感觉这个repo的hls代码明显领先于pynq脚本：pynq脚本的提交停留在了5years ago。

我现在能想到的两个思路是
- 使用master 分支进行复现。master branch整体停留在6年前。他的hls和pynq脚本应该是匹配的，同时bin权重也不同
- 自己手动修改pynq脚本的偏移地址以适配s axi的地址映射,也就是这一块代码：
```#===============================================
#                  yolov2 hw Ex
#==============================================

#IP_base_address
IP_BASE_ADDRESS    =  0x43C00000
ADDRESS_RANGE      = 0x180

XYOLO2_FPGA_CTRL_BUS_ADDR_AP_CTRL            =0x000
XYOLO2_FPGA_CTRL_BUS_ADDR_GIE                =0x004
XYOLO2_FPGA_CTRL_BUS_ADDR_IER                =0x008
XYOLO2_FPGA_CTRL_BUS_ADDR_ISR                =0x00c
XYOLO2_FPGA_CTRL_BUS_ADDR_INPUT_R_DATA       =0x010
XYOLO2_FPGA_CTRL_BUS_ADDR_INPUT1_DATA        =0x018
XYOLO2_FPGA_CTRL_BUS_ADDR_INPUT2_DATA        =0x020
XYOLO2_FPGA_CTRL_BUS_ADDR_INPUT3_DATA        =0x028
XYOLO2_FPGA_CTRL_BUS_ADDR_OUTPUT_R_DATA      =0x030
XYOLO2_FPGA_CTRL_BUS_ADDR_OUTPUT1_DATA       =0x038
XYOLO2_FPGA_CTRL_BUS_ADDR_WEIGHT_DATA        =0x040
XYOLO2_FPGA_CTRL_BUS_ADDR_BETA_DATA          =0x048
XYOLO2_FPGA_CTRL_BUS_ADDR_INFM_NUM_DATA      =0x050
XYOLO2_FPGA_CTRL_BUS_ADDR_OUTFM_NUM_DATA     =0x058
XYOLO2_FPGA_CTRL_BUS_ADDR_KERNEL_SIZE_DATA   =0x060
XYOLO2_FPGA_CTRL_BUS_ADDR_KERNEL_STRIDE_DATA =0x068
XYOLO2_FPGA_CTRL_BUS_ADDR_INPUT_W_DATA       =0x070
XYOLO2_FPGA_CTRL_BUS_ADDR_INPUT_H_DATA       =0x078
XYOLO2_FPGA_CTRL_BUS_ADDR_OUTPUT_W_DATA      =0x080
XYOLO2_FPGA_CTRL_BUS_ADDR_OUTPUT_H_DATA      =0x088
XYOLO2_FPGA_CTRL_BUS_ADDR_PADDING_DATA       =0x090
XYOLO2_FPGA_CTRL_BUS_ADDR_ISNL_DATA          =0x098
XYOLO2_FPGA_CTRL_BUS_ADDR_ISBN_DATA          =0x0a0
XYOLO2_FPGA_CTRL_BUS_ADDR_TM_DATA            =0x0a8
XYOLO2_FPGA_CTRL_BUS_ADDR_TN_DATA            =0x0b0
XYOLO2_FPGA_CTRL_BUS_ADDR_TR_DATA            =0x0b8
XYOLO2_FPGA_CTRL_BUS_ADDR_TC_DATA            =0x0c0
XYOLO2_FPGA_CTRL_BUS_ADDR_MLOOPS_DATA        =0x0c8
XYOLO2_FPGA_CTRL_BUS_ADDR_NLOOPS_DATA        =0x0d0
XYOLO2_FPGA_CTRL_BUS_ADDR_RLOOPS_DATA        =0x0d8
XYOLO2_FPGA_CTRL_BUS_ADDR_CLOOPS_DATA        =0x0e0
XYOLO2_FPGA_CTRL_BUS_ADDR_LAYERTYPE_DATA     =0x0e8
XYOLO2_FPGA_CTRL_BUS_ADDR_INPUTQ_DATA        =0x0f0
XYOLO2_FPGA_CTRL_BUS_ADDR_OUTPUTQ_DATA       =0x0f8
XYOLO2_FPGA_CTRL_BUS_ADDR_WEIGHTQ_DATA       =0x100
XYOLO2_FPGA_CTRL_BUS_ADDR_BETAQ_DATA         =0x108
XYOLO2_FPGA_CTRL_BUS_ADDR_TROW_LOOPS_DATA    =0x110

#XYOLO2_FPGA_CTRL_BUS_ADDR Write data

def YOLO__Init_EX(In_Address,Out_Address,Weight_offset,Beta_offset,InFM_num,OutFM_num,
                 Kernel_size,Kernel_stride,
                 Input_w,Input_h,Output_w,Output_h,
                  Padding,IsNL,IsBN,
                 TM,TN,TR,TC,
                 mLoops,nLoops,rLoops,cLoops,LayerType,
                 InputQ,OutputQ,WeightQ,BetaQ,WEIGHT_BASE,BETA_BASE):
```

pgq18, 2024-01-11:
> 目前我没能解决这个问题，这个仓库的复现在我这搁置了。 我感觉这个repo的hls代码明显领先于pynq脚本：pynq脚本的提交停留在了5years ago。
>
> 我现在能想到的两个思路是
>
> * 使用master 分支进行复现。master branch整体停留在6年前。他的hls和pynq脚本应该是匹配的，同时bin权重也不同
> * 自己手动修改pynq脚本的偏移地址以适配s [...]

好的谢谢，我现在在pynq-z1上复现出来了，用的是100MHzTn2Tm32Tr26Tc26这个分支里的内容，然后把ipynb里的地址按照我综合出来的地址进行更改，就可以运行了。谢谢。

gb16001, 2024-01-14:
出现新bug，自己综合的IP**没有输出或给一个固定的错误输出**。
### 复现步骤
我使用100MHzTn2Tm32Tr26Tc26分支中的内容进行hls,overlay及py脚本测试。权重也使用这个branch中的
`"yolov2_w_reorg_bn_ap16_short16.bin","yolov2_b_ap16_short16.bin"`
- 使用分支中发布的bit file进行测试
一切正常（用`from pynq import allocate `替换了xlin库，因为新pynq系统镜像不再支持）
- 使用自己综合的IP测试
修正IP驱动偏移地址后，程序可以正常运行。
修改后的ip drive
```
######self fixed drive
#===============================================
#                  yolov2 hw Ex
#==============================================

#IP_base_address
IP_BASE_ADDRESS    =  0x43C00000
ADDRESS_RANGE      = 4000
#XYOLO2_FPGA_CTRL_BUS_ADDR 
XYOLO2_FPGA_CTRL_BUS_ADDR_AP_CTRL            =0x000
XYOLO2_FPGA_CTRL_BUS_ADDR_GIE                =0x004
XYOLO2_FPGA_CTRL_BUS_ADDR_IER                =0x008
XYOLO2_FPGA_CTRL_BUS_ADDR_ISR                =0x00c
XYOLO2_FPGA_CTRL_BUS_ADDR_INPUT_R_DATA       =0x010
XYOLO2_FPGA_CTRL_BUS_ADDR_OUTPUT_R_DATA      =0x01c
XYOLO2_FPGA_CTRL_BUS_ADDR_WEIGHT_DATA        =0x028
XYOLO2_FPGA_CTRL_BUS_ADDR_BETA_DATA          =0x034
XYOLO2_FPGA_CTRL_BUS_ADDR_INFM_NUM_DATA      =0x040
XYOLO2_FPGA_CTRL_BUS_ADDR_OUTFM_NUM_DATA     =0x048
XYOLO2_FPGA_CTRL_BUS_ADDR_KERNEL_SIZE_DATA   =0x050
XYOLO2_FPGA_CTRL_BUS_ADDR_KERNEL_STRIDE_DATA =0x058
XYOLO2_FPGA_CTRL_BUS_ADDR_INPUT_W_DATA       =0x060
XYOLO2_FPGA_CTRL_BUS_ADDR_INPUT_H_DATA       =0x068
XYOLO2_FPGA_CTRL_BUS_ADDR_PADDING_DATA       =0x070
XYOLO2_FPGA_CTRL_BUS_ADDR_ISNL_DATA          =0x078
XYOLO2_FPGA_CTRL_BUS_ADDR_ISBN_DATA          =0x080
XYOLO2_FPGA_CTRL_BUS_ADDR_TM_DATA            =0x088
XYOLO2_FPGA_CTRL_BUS_ADDR_TN_DATA            =0x090
XYOLO2_FPGA_CTRL_BUS_ADDR_TR_DATA            =0x098
XYOLO2_FPGA_CTRL_BUS_ADDR_TC_DATA            =0x0a0
XYOLO2_FPGA_CTRL_BUS_ADDR_MLOOPS_DATA        =0x0a8
XYOLO2_FPGA_CTRL_BUS_ADDR_NLOOPS_DATA        =0x0b0
XYOLO2_FPGA_CTRL_BUS_ADDR_RLOOPS_DATA        =0x0b8
XYOLO2_FPGA_CTRL_BUS_ADDR_CLOOPS_DATA        =0x0c0
XYOLO2_FPGA_CTRL_BUS_ADDR_LAYERTYPE_DATA     =0x0c8
XYOLO2_FPGA_CTRL_BUS_ADDR_INPUTQ_DATA        =0x0d0
XYOLO2_FPGA_CTRL_BUS_ADDR_OUTPUTQ_DATA       =0x0d8
XYOLO2_FPGA_CTRL_BUS_ADDR_WEIGHTQ_DATA       =0x0e0
XYOLO2_FPGA_CTRL_BUS_ADDR_BETAQ_DATA         =0x0e8

def YOLO__Init_EX(In_Address,Out_Address,Weight_offset,Beta_offset,InFM_num,OutFM_num,
                 Kernel_size,Kernel_stride,
                 Input_w,Input_h,Padding,IsNL,IsBN,
                 TM,TN,TR,TC,
                 mLoops,nLoops,rLoops,cLoops,LayerType,
                 InputQ,OutputQ,WeightQ,BetaQ,WEIGHT_BASE,BETA_BASE):
   
    # mapping memory
#     mmio = MMIO(IP_BASE_ADDRESS,ADDRESS_RANGE)
    mmio=overlay.YOLO2_FPGA_0.mmio # new driver of mmio of yolo2_fpga ip core
    
    
    while True:
        ap_idle =  (mmio.read(XYOLO2_FPGA_CTRL_BUS_ADDR_AP_CTRL)>>2)&0x01
        if(ap_idle):
            break
            
            
    mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_INPUT_R_DATA,  In_Address)
    mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_OUTPUT_R_DATA, Out_Address)
    mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_WEIGHT_DATA,   WEIGHT_BASE+Weight_offset*4)
    mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_BETA_DATA,     BETA_BASE+Beta_offset*4)

    mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_INFM_NUM_DATA, InFM_num)
    mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_OUTFM_NUM_DATA, OutFM_num)
    mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_KERNEL_SIZE_DATA, Kernel_size)
    mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_KERNEL_STRIDE_DATA, Kernel_stride)
    mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_INPUT_W_DATA, Input_w)
    mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_INPUT_H_DATA, Input_h)
    mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_PADDING_DATA, Padding)
    mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_ISNL_DATA, IsNL)
    mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_ISBN_DATA, IsBN)
    mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_TM_DATA, TM)
    mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_TN_DATA, TN)
    mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_TR_DATA, TR)
    mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_TC_DATA, TC)
    mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_MLOOPS_DATA, mLoops)
    mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_NLOOPS_DATA, nLoops)
    mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_RLOOPS_DATA, rLoops)
    mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_CLOOPS_DATA, cLoops)
    mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_LAYERTYPE_DATA, LayerType)
    mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_INPUTQ_DATA, InputQ)
    mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_OUTPUTQ_DATA, OutputQ)
    mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_WEIGHTQ_DATA, WeightQ)
#   mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_ALPHAQ_DATA, AlphaQ)
    mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_BETAQ_DATA, BetaQ)
    mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_GIE,0)
    mmio.write(XYOLO2_FPGA_CTRL_BUS_ADDR_AP_CTRL,1)
    while True:
        ap_done =  (mmio.read(XYOLO2_FPGA_CTRL_BUS_ADDR_AP_CTRL)>>1)&0x01
        if(ap_done):
            break
```

但推理结果没有识别框或给一组固定的错误识别。且运行时间偏长
`self:6.024651527404785>>released:2.6518325805664062` 
[image]

### 问题
感觉这个bug与[ issue#98](https://github.com/dhm2013724/yolov2_xilinx_fpga/issues/98)描述的很像。请问你碰到了类似的问题吗？有没有解决？
### bug原因猜测
个人感觉是权重文件和py脚本中模型配置参数的问题。hls IP只实现层计算，层间结构由py脚本加载bin权重实现。感觉是没有正确组织起网络结构，用IP实现的计算层。因为release IP使用flex branch bin也出现了乱输出的情况。（而且detection头，即最后一层，的实现好像也是在py脚本中实现）
也有可能是因为overlay实现的bit file 根本就没有实现层计算功能，IP根本就没工作

pgq18, 2024-01-14:
> 出现新bug，自己综合的IP**没有输出或给一个固定的错误输出**。
>
> ### 复现步骤
> 我使用100MHzTn2Tm32Tr26Tc26分支中的内容进行hls,overlay及py脚本测试。权重也使用这个branch中的 `"yolov2_w_reorg_bn_ap16_short16.bin","yolov2_b_ap16_short16.bin"`
>
> * 使用分支中发布的bit [...]

我没有遇到这个问题，但是我猜测是不是跟hls综合时的设置有关系，比如说device，estimated period的设定。这个是我的设置，我用的是pynq-z1。
[image]

pgq18, 2024-01-14:
我没有遇到这个问题，但是我猜测是不是跟hls综合时的设置有关系，比如说device，estimated period的设定。这个是我的设置，我用的是pynq-z1，是根据分支里面hls文件夹里的script.tcl进行设置的。
[image]

gb16001, 2024-01-15:
问题已解决，是因为vivado overlay 中 ps IP的hp0被设置为了32位而不是64位。改为64位后一切正常。
[image]

### debug过程简述
整个复现过程中，复现版与release版的区别只有bit file and ip drive。所以一定是它们两中出了问题。为了检查bit file 中的IP是否工作正常，将其推理结果导出进行查看。发现于release版的推理结果有较大差距，进而锁定错误。
#### a.检查py脚本中fpga的推理解结果
添加代码将fpga的推理结果region_buff 导出
```
    #fpga_process_time start
    start_time = time.time()
    region_buff = np.zeros((73008,), dtype=np.float32)
  
    yolo_fpga(img_base_buffer,region_buff)
    end_time = time.time()
    fpga_process_time = end_time - start_time 
    #fpga_process_time end
    if debug_fpga_result:
        if 'batch_code' in globals():
            fpga_result_file_name=f"infer_result/fpga_result_{os.path.splitext(os.path.basename(path))[0]}_{ORIG_IMG_PATH[:-4]}_{batch_code}.txt" 
        else:
            fpga_result_file_name=f"infer_result/fpga_result_{os.path.splitext(os.path.basename(path))[0]}_{ORIG_IMG_PATH[:-4]}.txt" 
            pass
        
        print("saving infer result to:",fpga_result_file_name)
        np.savetxt(fpga_result_file_name, region_buff)
        del fpga_result_file_name
        pass
```
**现象**
```
# 部分输出
-5.924544900000000000e+07
0.000000000000000000e+00
-5.924544900000000000e+07
0.000000000000000000e+00
-5.924544900000000000e+07
0.000000000000000000e+00
-5.924544900000000000e+07
0.000000000000000000e+00
```
大量重复的数字中间夹杂着空行，查明release bit file的正常推理没有这些空行。故锁定为bit file中的IP位工作正常。

closed 2024-01-15 by gb16001